esphome/feature-requests #1996: feature request for an 'update all' button

roblatour, 2022-12-07:
**Describe the problem you have/What new integration you would like**
When updating the core and/or multiple add ons via Settings if there is more that one thing to update, rather than have to install each update one after another add a button that says 'update all' that when pressed will everything one after another

**Please describe your use case for this integration and alternatives you've tried:**
currently i need to install everything one by one; this takes my time.  Would be way better to have an 'update all' button and I could come back later to find it all done: -)

**Additional context**

nagyrobi, 2022-12-07:
If you're referring to ESPHome, the _Update All_ button exists in the dashboard:
[image]

If you're referring to Home Assistant, you've submitted your request in the wrong place.

closed 2022-12-07 by nagyrobi

roblatour, 2022-12-07:
Please reopen this.
I am not referring to ESP Home.
I am referring to when the Settings entry in the left panel of the main window show that there are updates to be performed for example it can say Settings with a 3 in red beside it, then when you go into the next window to install those updates you currently need to do them one at a time.

nagyrobi, 2022-12-07:
This repository is for holding feature requests only for ESPHome.
Your request relates to Home Assistant, please re-post it in the correct place: https://community.home-assistant.io/c/feature-requests/13

roblatour, 2022-12-07:
@nagyrobi thank you - I'll give it a try