esphome/feature-requests #1997: Scaling for number component

prigorus, 2022-12-07:
**Describe the problem you have/What new integration you would like**

The device sends values that are scaled by 10. I want to write these values to a number (to display in the frontend and change) and then send them back using the same number. The number component doesn't allow scaling like the sensor component. 

**Please describe your use case for this integration and alternatives you've tried:**

There is the option to write the number to a global variable and display it as a sensor template, but this creates two entities of the same number. If the number component would have a scaling option, it would make the code much cleaner and easier to use. 
This, in my opinion is a very common feature for others that are using Tuya MCU or something similar where values are multiplied to only use integers instead of float. 

**Additional context**

ssieb, 2022-12-07:
I would consider this to be an issue actually, but I'll leave it here for now.  It should work similarly to the Tuya climate component where there's a multiplier for the temperature sensor.

prigorus, 2023-01-26:
Is there any option to get this implemented? Should I close the feature request and submit a bug instead?

Thank you!

prigorus, 2023-03-16:
Opened issue instead

closed 2023-03-16 by prigorus