esphome/feature-requests #2000: Allow option to compile yaml only

pedjas, 2022-12-07:
**Describe the problem you have/What new integration you would like**

It would be great to have separate Compile button that would only do syntax check and compilation without uploading to device.

**Please describe your use case for this integration and alternatives you've tried:**

While writing code I have habit  of frequently compile code to make sure it is OK without intention to actually upload it to device. I am pretty sure I am not the only one that does this way.

There is no such option. The only way to compile from editor is to run Install which goes all way to upload.

**Additional context**

pedjas, 2022-12-07:
> I sometimes do that too, but press STOP to cancel, right before upload process.

Than I end up with other issue: if I press STOP I loose compile log.

bkaufx, 2022-12-07:
Can you just install with manual download? That's what I do all the time when I don't actually want to upload to the device.