esphome/feature-requests #2003: Customizing order of components in web interface

pedjas, 2022-12-07:
**Describe the problem you have/What new integration you would like**

When setting up yaml it would be good if user can set order of components how to show up in web interface.

**Please describe your use case for this integration and alternatives you've tried:**

No matter what order of components user use in yaml, web interface shows them in alphabetical order. That usually is not best choice.

It seems the only way to achieve this is to add some prefixes to component names, which just makes names longer and unnecessarily clutters web interface.

It would be good for user to have option to set order of components for user interface which is more logical, natural and understandable for end user.

**Additional context**

nagyrobi, 2022-12-07:
Related to https://github.com/esphome/feature-requests/issues/1620

randyoo, 2024-09-01:
This has been implemented in V3 via "web_server_sorting_weight:" property.

closed 2024-09-01 by nagyrobi