esphome/feature-requests #2004: Option to automaticaly create and install minimal configuration to clear space in device

pedjas, 2022-12-07:
**Describe the problem you have/What new integration you would like**

When editing configuration within ESPHome and installing via web, it quite often happens that uploader reports there is no enough space on device for upload. 

Solution is to cut of all unnecessary content form yaml, copy it somewhere as it is needed later, upload minimal configuration to clear space and then return back all cut content in yaml and upload full version.

It happens quite frequently and becomes annoying.

It would be great if there is option to install minimal configuration that would take only necessary configuration from current yaml to create minimal configuration and upload that without need for user to manually cut things out.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

brammeleman, 2022-12-15:
So, what's the minimal configuration? One that has just wifi and ota? Ie:

``` yaml
esphome:
  name: awesome
esp8266:
  board: d1_mini
ota:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
Firmware size 311K

related #1264 #1499

pedjas, 2022-12-17:
Yes, what is necessary to have OTA functioning. In most cases it can fit available space.