esphome/feature-requests #2008: Adding static host entries (/etc/hosts capability)

segdy, 2022-12-08:
**Describe the problem you have/What new integration you would like**

Ability to add a host entry to ESPhome that will be used to resolve arbitrary DNS names locally without any DNS required.

This is standard functionality of any TCP/IP stack and it would be very useful to have this in ESPhome as well.

**Please describe your use case for this integration and alternatives you've tried:**

I have [asked](https://github.com/esphome/esphome/discussions/4050) this but since there was no response I assume it it not possible (yet).

The use cases are manifold. For example, you may want to operate your ESP devices without relying on a DNS server. Or you may want to put your ESP devices into a dedicated VLAN without DNS.

**Additional context**

Currently I am not able to use the media player at all. Reason: The requesting system (Home Assistant) sends a request to play a file to my ESPhome media player. This file is sent in the form of an URL. The URL points back to my Home Assistant but unfortunately the DNS name it uses is not configurable. On any other system I would just enter something like this into my /etc/hosts file:

    10.227.68.10   home-assistant.intra.example.net

nagyrobi, 2022-12-15:
In the [core network component](https://esphome.io/components/network.html) I'd imagine something like:

```
# Example configuration
network:
    enable_ipv6: true
    hosts:
      - 10.227.68.10:   home-assistant.intra.example.net
      - 10.227.68.11:   home-assistant-test.intra.example.net
```

segdy, 2022-12-15:
Yes that would be exactly what I had been imagining as well.

nagyrobi, 2022-12-15:
Afaik we're using `lwip` which doesn't seem to support it: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html