esphome/feature-requests #2010: Support Tuya BLE/MESH device

Jarled, 2022-12-09:
**Describe the problem you have/What new integration you would like**

How to add Tuya BLE Sensors 
Temp & Humidity Sensor, Model: TH05 BLE device

**Please describe your use case for this integration and alternatives you've tried:**

[Tuya TH05](https://pvvx.github.io/TH-05/) is very similar to [LYWSD03MMC](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi#lywsd03mmc) so would expect the Tuya version to also work in the same way. i have been able to get the details from the Tuya Dev portal so i guess the BLE bind key is the "local key"

> Office Temperature & Humidity Sensor
> {
>   "result": {
>     "active_time": 1668864724,
>     "category": "wsdcg",
>     "category_name": "Temperature and Humidity Sensor",
>     "gateway_id": "",
>     "icon": "smart/icon/ay1525749833414yotNt/bfd21e12a1179860998fb0bdec6a048d.png",
>     "id": "bf8918ypm9******",
>     "local_key": "1126266b1858c582",
>     "model": "",
>     "name": "Office Temperature & Humidity Sensor",
>     "online": false,
>     "owner_id": "66942552",
>     "product_id": "vyfoip9h",
>     "product_name": "Temp & Humidity Sensor",
>     "sub": false,
>     "time_zone": "+00:00",
>     "update_time": 1668864762,
>     "uuid": "2bb3cf444ac24cee"
>   },
>   "success": true,
>   "t": 1668978709007,
>   "tid": "f25b3363691711ed8c8c06bdf2add754"
> }
> 
> ------------------------------------------------
> Temperature & Humidity Sensor
> {
>   "result": {
>     "active_time": 1668864584,
>     "category": "wsdcg",
>     "category_name": "Temperature and Humidity Sensor",
>     "icon": "smart/icon/ay1525749833414yotNt/bfd21e12a1179860998fb0bdec6a048d.png",
>     "id": "bfd7fcmjac******",
>     "local_key": "dc5df19ff7843376",
>     "model": "",
>     "name": "Bedroom Temperature & Humidity Sensor",
>     "online": false,
>     "owner_id": "66942552",
>     "product_id": "vyfoip9h",
>     "product_name": "Temp & Humidity Sensor",
>     "sub": false,
>     "time_zone": "+00:00",
>     "update_time": 1668864679,
>     "uuid": "a2344189efe61b5f"
>   },
>   "success": true,
>   "t": 1668978787680,
>   "tid": "213fb754691811ed81fb8ec426404b74"
> }

**Additional context**

Could you add this

Jarled, 2022-12-09:
#https://github.com/pvvx/ATC_MiThermometer/issues/268

edudele, 2023-09-11:
@Jarled Did you achieve something or did you give up?

Jarled, 2023-09-12:
Gave up