esphome/feature-requests #2012: Syntax coloring in lambdas

pedjas, 2022-12-10:
**Describe the problem you have/What new integration you would like**

Code is lambdas should be syntax colored.

**Please describe your use case for this integration and alternatives you've tried:**

It is not rare situation when user has to use lambda. ESPHome editor does not do syntax coloring in lambda so content is not easy to follow, and some typing errors are hard to notice.

Coloring would really help.

**Additional context**

bkaufx, 2022-12-11:
Weird thing is it works in the VS Code plugin and I swear it used to work in the dashboard too.

clowrey, 2024-11-08:
I would love to see syntax highlighting for the lambdas too!
Here is with the current ESPhome dashboard v2024.10.3:
[image]

With the current VSCode ESPhome extension v2023.9.0: 
[image]

closed 2025-04-14 by jesserockz