esphome/feature-requests #2013: Add pairing support to BTProxy

dbuezas, 2022-12-11:
**Describe the problem you have/What new integration you would like**

To be able to pair with BLE devices from the BTProxy, even if the pairing pin has to be entered via YAML.

**Please describe your use case for this integration and alternatives you've tried:**

The Bluetooth EQ3 Thermostats integration require active and paired connections and they stop working as soon as a BTProxy has better RSSI than the local adapter.

**Additional context**

This is the custom EQ3 Thermostat integration I am using: https://github.com/dbuezas/eq3btsmart
The integration in HA's core has been broken for a couple of months so I made this fork that brings back support, adds discovery and config-flows, all missing entities and services to change the internal heating schedules. I intend to publish the new python library and make a PR to the HA core once the incompatibility with BTProxy is solved.

I gave it a shot to implement pairing in esphome but did not succeed. I understand HA doesn't support it yet, but adding pairing pin via yaml would already be super useful in these cases :)

bdraco, 2022-12-18:
bleak is adding pairing APIs which should make this possible in the future see https://github.com/hbldh/bleak/pull/1100

G-S-E, 2022-12-19:
> bleak is adding pairing APIs which should make this possible in the future see [hbldh/bleak#1100](https://github.com/hbldh/bleak/pull/1100)

Why do you think that this is going to work with bluetooth_proxy?? I don’t think so as the pairing is done by particular esp32. Furthermore, many devices cam be paired only with one master at the same time… You cannot pair with device directly via bleak when you use bt proxy and then connect to it via another proxy or bt adapter because the key exchange is done between the paired devices only…

dbuezas, 2022-12-19:
From the integration perspective, BTProxys don't exist. HA manages the routing so I think he means that once the API is there the ESPHome team will be unblocked from implementing the proxy side. And who knows, maybe the keys can be shared like [this](https://www.castoriscausa.com/posts/2021/02/28/bluetooth-dual-boot/)

G-S-E, 2022-12-21:
> And who knows, maybe the keys can be shared like [this](https://www.castoriscausa.com/posts/2021/02/28/bluetooth-dual-boot/)

This case is different because in both OS the MAC is the same, so you can share the key. In our case with proxies MACs are different

Eroli, 2023-01-11:
> bleak is adding pairing APIs which should make this possible in the future see [hbldh/bleak#1100](https://github.com/hbldh/bleak/pull/1100)

Do I understand it correctly that the lack of pairing API in Bleak is the problem?
I have already worked with the ESP32 and the NimBLE Arduino Library with the EQ3 thermostats and that worked fine. Maybe an integration in HA/ESPHome via NimBLE-Arduino would be possible?

https://github.com/h2zero/NimBLE-Arduino

ignisf, 2023-01-21:
https://github.com/esphome/esphome-docs/pull/2607

milandzuris, 2025-10-16:
something new?