esphome/feature-requests #2017: Support for SSD1306 variant: 72x40 pixels

kquinsland, 2022-12-13:
**Describe the problem you have/What new integration you would like**

I am asking for an expansion to an **existing** integration.
The SSD1306 integration already has several displays that are _close_ in size, but are not quite correct.

**I am asking for the `SSD1306` component to support `72x40` resolution**

**Please describe your use case for this integration and alternatives you've tried:**

I am currently trying a `display:` component with `SSD1306 64x48` but the results are ... illegible.
Pixels are lighting up but they're not forming the image I'd expect (just `printf('HI')`).

The display doing _anything_ at all makes sense as it's a common family of controllers.
The "corrupted" image would also make sense if the text is being rendered into a frame buffer that does not match the resolution of my display.

**Additional context**

This is the ESP32/C3 module that I am using / first discovered that this display is even a thing:

https://github.com/01Space/ESP32-C3-0.42LCD

I was able to figure out that it's a common SSD1306 type display by consulting the source code that the module ships with: https://github.com/01Space/ESP32-C3-0.42LCD/blob/main/ESP32-C3-GraphicsTest/ESP32-C3-GraphicsTest.ino#L109

You can find various sellers online with the display in "breakout" form: https://www.buydisplay.com/i2c-white-0-42-inch-oled-display-module-72x40-arduino-raspberry-pi

romainrossi, 2023-02-11:
+1 Same board, same problem

ezpl, 2023-03-11:
I have the board working using in PlatformIO and the U8G2 library. I would imagine it would be firly easy to add support for this display into ESPHome. 
I can find this code [ssd1306_base.h](https://esphome.io/api/ssd1306__base_8h_source.html) but surely there must be more to it than just updating the enum? Would be great if someone could point me in the right direction / some instructions on how to get started.

johnsto, 2023-04-13:
https://github.com/esphome/esphome/pull/4659