esphome/feature-requests #2021: [Fingerprint_Grow] Sync Fingerprints Between Sensors

Michael-Loia, 2022-12-16:
**Describe the problem you have/What new integration you would like**

Existing integration, "fingerprint_grow"

**Please describe your use case for this integration and alternatives you've tried:**

I haven't be able to find any alternatives, however, I have the general solution.

The proposed feature for the fingerprint sensors would allow for a system in which a master sensor and multiple slave sensors can sync their fingerprints.

The master sensor would store the enrolled fingerprints, and then send the fingerprints to all of the slave sensors. The slave sensors would then store those fingerprints, so that all of the sensors are in sync.

When the master sensor enrolls a new fingerprint, it will collect all of the saved fingerprints and send them to the slave sensors, overwriting whatever is already saved.

[image]

**Additional context**

There would be a setting for master sensor in the configuration that would look something like the following
```
fingerprint_grow:
  sync_enabled: true
  sync_master: true
  sensing_pin: GPIO4
```
New option: "**sync_enabled**", optional, defaults to "**false**"
New option: "**sync_master**", optional, defaults to "**false**"
If master is set to true then it would send its fingerprints to all sensors that are not, under the condition that sync is enabled

Perhaps over mqtt, the master sensor would send it's configuration over mqtt (maybe as json) and slave sensors would be subscribed to it with qos 2

[image]

Most ideally it would delete all fingerprints on the slave sensor before importing from the master sensor, to prevent any sync issues.
Why? - If master has 5 fingerprints saved on it but slave has 7, the 5 would be replaced but the 2 would be left alone because there was no replacement sent.
This would prevent issues like that.

==========

It looks like there is a section where it saves the fingerprint to the sensor (Between Lines 161 and 187)
https://esphome.io/api/fingerprint__grow_8cpp_source.html

It looks like it should be possible to send the final results after the event of saving / enrolling based on this snippet of code from the link

```
 uint8_t FingerprintGrowComponent::save_fingerprint_() {
   ESP_LOGI(TAG, "Creating model");
   this->data_ = {REG_MODEL};
   switch (this->send_command_()) {
     case OK:
       break;
     case ENROLL_MISMATCH:
       ESP_LOGE(TAG, "Scans do not match");
     default:
       return this->data_[0];
   }
 
   ESP_LOGI(TAG, "Storing model");
   this->data_ = {STORE, 0x01, (uint8_t)(this->enrollment_slot_ >> 8), (uint8_t)(this->enrollment_slot_ & 0xFF)};
   switch (this->send_command_()) {
     case OK:
       ESP_LOGI(TAG, "Stored model");
```
---->  // Add code to get all models (fingerprints I think?) and send them over mqtt to all of the slave sensors
```
       break;
     case BAD_LOCATION:
       ESP_LOGE(TAG, "Invalid slot");
       break;
     case FLASH_ERR:
       ESP_LOGE(TAG, "Error writing to flash");
       break;
   }
   return this->data_[0];
 }
```

Example JSON:
```
{
"0": {"data": "uint8_t string"},
"1": {"data": "uint8_t string"},
"2": {"data": "uint8_t string"},
"3": {"data": "uint8_t string"},
"4": {"data": "uint8_t string"}
}
```
This example shows 5 fingerprints that would be sent from the master sensor to the slave sensors, with the data of each being the uint8_t string that fingerprint is saved as.

It looks like this part shows code that we can use to get all of the fingerprints (Between Lines 90 and 119)
https://esphome.io/api/fingerprint__grow_8cpp_source.html

```
 void FingerprintGrowComponent::scan_and_match_() {
   if (this->sensing_pin_ != nullptr) {
     ESP_LOGD(TAG, "Scan and match");
   } else {
     ESP_LOGV(TAG, "Scan and match");
   }
   if (this->scan_image_(1) == OK) {
     this->waiting_removal_ = true;
     this->data_ = {SEARCH, 0x01, 0x00, 0x00, (uint8_t)(this->capacity_ >> 8), (uint8_t)(this->capacity_ & 0xFF)};
     switch (this->send_command_()) {
       case OK: {
         ESP_LOGD(TAG, "Fingerprint matched");
         uint16_t finger_id = ((uint16_t) this->data_[1] << 8) | this->data_[2];
         uint16_t confidence = ((uint16_t) this->data_[3] << 8) | this->data_[4];
         if (this->last_finger_id_sensor_ != nullptr) {
           this->last_finger_id_sensor_->publish_state(finger_id);
         }
         if (this->last_confidence_sensor_ != nullptr) {
           this->last_confidence_sensor_->publish_state(confidence);
         }
         this->finger_scan_matched_callback_.call(finger_id, confidence);
         break;
       }
       case NOT_FOUND:
         ESP_LOGD(TAG, "Fingerprint not matched to any saved slots");
         this->finger_scan_unmatched_callback_.call();
         break;
     }
   }
 }
```

Thank you for your time, and consideration. Please let me know if I can do anything to help.

Best,
Michael

nagyrobi, 2022-12-16:
Does the device permit extracting the fingerprint codes? And allow writing it?

Michael-Loia, 2022-12-16:
Hi nagyrobi, thank you for your prompt response.

Based on line 98, it appears the code reads all the saved fingerprints. That specific use case is comparing a lively scanned fingerprint to a saved one. I would assume that we would be able to access that without scanning a fingerprint.
https://esphome.io/api/fingerprint__grow_8cpp_source.html

I've found no way to manually test this, if you could point me in the right direction I would be more than happy to!