esphome/feature-requests #2022: Using Bluetooth buttons that indentify themself as keyboard (AB Shutter3)

chanz, 2022-12-18:
**Describe the problem you have/What new integration you would like**

For better readability:
- _AB Shutter3_ is the Bluetooth name of the button that I bought [here](https://www.amazon.de/gp/product/B07ZG291YX/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1).
- _BLE iTag Button_ is the Bluetooth button mentioned in the example [here](https://esphome.io/cookbook/ble_itag.html).

I'd like to use Bluetooth keyboards or Bluetooth buttons, to work with esphome integrations and Home Assistant. The Bluetooth button _AB Shutter3_ I bought, registers itself as keyboard and sends "volume up" to the connected computer/smartphone.

**Please describe your use case for this integration and alternatives you've tried:**
 
My use cases:
- I like to use the _AB Shutter3_ as a cheap alternative to toggle lights on and off.
- _Optional:_ Using Bluetooth keyboards, stickers/labels and a wall mount to create an terminal that controls every part of my home automations without the use of submenus or key combinations.

Alternatives I've tried:
- I started with [this](https://esphome.io/cookbook/ble_itag.html) example. Since pressing the button on _AB Shutter3_ did not provide any output, I tried to understand how Bluetooth works. My guess is that the _BLE iTag Button_ supports notifications for the button press characteristics and the _AB Shutter3_ does not.
**Notice:** The battery service and characteristic of the _AB Shutter3_ did work and shows 99%.
- Using the service and characteristic list of _AB Shutter3_ (see below), I tried every service and characteristic UUID, but sadly nothing worked.
- I've searched how to connect Bluetooth keyboards to an esp32 and found [this](https://github.com/esp32beans/BLE_HID_Client/blob/main/examples/BLE_HID_Client/BLE_HID_Client.ino). But I lack the knowledge to make use of it for esphome.

**Additional context**

- This file provides a tcpdump where I recorded the paring and pressing the button of the _AB Shutter3_ two times with 5 seconds in between. [bt_capture_ab-shutter3_pair-and-button.pcap.txt](https://github.com/esphome/feature-requests/files/10254306/bt_capture_ab-shutter3_pair-and-button.pcap.txt)
- The following was my best guess to make the _AB Shutter3_ work in esphome:
```
esphome:
  name: testbench
  platform: ESP32
  board: esp-wrover-kit

# Enable logging
logger:
  #level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "<removed>"

wifi:
  ssid: "<removed>"
  password: "<removed>"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testbench Fallback Hotspot"
    password: "<removed>"

captive_portal:

esp32_ble_tracker:

ble_client:
  # Replace with the MAC address of your device.
  - mac_address: 2A:07:98:10:37:05
    id: bt_client_black_button_1

binary_sensor:
  - platform: template
    id: bt_black_button_1
    name: "BT - Black Button 1"
    filters:
      delayed_off: 200ms

sensor:
# This entry registers and awaits notifications for the
# characteristic that signals button presses. Each time
# a notification is received, the corresponding binary_sensor
# is briefly toggled.
  - platform: ble_client
    type: characteristic
    ble_client_id: bt_client_black_button_1
    name: "BT - Black Button 1 - Switch"
    service_uuid: '1812'
    characteristic_uuid: '2A4C'
    notify: true
    update_interval: never
    on_notify:
      then:
        - binary_sensor.template.publish:
            id: bt_black_button_1
            state: ON
        - binary_sensor.template.publish:
            id: bt_black_button_1
            state: OFF

# This entry queries the battery level. Some tags may not
# support this characteristic, you will see 'Unknown' in the
# HA frontend.
  - platform: ble_client
    type: characteristic
    ble_client_id: bt_client_black_button_1
    name: "BT - Black Button 1 - Battery"
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    icon: 'mdi:battery'
    unit_of_measurement: '%'

  - platform: ble_client
    type: rssi
    ble_client_id: bt_client_black_button_1
    name: "BT - Black Button 1 - RSSI"
```
- The following is the log of the esp32 I use within esphome:
```
[18:25:43][D][esp32_ble_client:048]: [0] [2A:07:98:10:37:05] Found device
[18:25:43][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[18:25:43][I][esp32_ble_client:064]: [0] [2A:07:98:10:37:05] 0x00 Attempting BLE connection
[18:25:43][I][ble_sensor:031]: [BT - Black Button 1 - Switch] Connected successfully!
[18:25:43][I][ble_sensor:031]: [BT - Black Button 1 - Battery] Connected successfully!
[18:25:43][I][ble_rssi_sensor:027]: [BT - Black Button 1 - RSSI] Connected successfully!
[18:25:44][D][esp32_ble_tracker:327]: Starting scan...
[18:25:55][D][sensor:127]: 'BT - Black Button 1 - Battery': Sending state 99.00000 % with 0 decimals of accuracy
[18:26:09][I][ble_rssi_sensor:052]: ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT RSSI: -90
[18:26:09][D][sensor:127]: 'BT - Black Button 1 - RSSI': Sending state -90.00000 dBm with 0 decimals of accuracy
[18:26:55][D][sensor:127]: 'BT - Black Button 1 - Battery': Sending state 99.00000 % with 0 decimals of accuracy
[18:27:09][I][ble_rssi_sensor:052]: ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT RSSI: -100
[18:27:09][D][sensor:127]: 'BT - Black Button 1 - RSSI': Sending state -100.00000 dBm with 0 decimals of accuracy
```
- This is the service and characteristics list of the _AB Shutter3_:
```
[01:13:17][I][ble_client:161]: Service UUID: 0x1801 									- Generic Attribute service
[01:13:17][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x4
[01:13:17][I][ble_client:347]:  characteristic 0x2A05, handle 0x3, properties 0x20      	- Service Changed
[01:13:17][I][ble_client:161]: Service UUID: 0x1800 									- Generic Access service
[01:13:17][I][ble_client:162]:   start_handle: 0x5  end_handle: 0xb
[01:13:17][I][ble_client:347]:  characteristic 0x2A00, handle 0x7, properties 0x2 			- Device Name
[01:13:17][I][ble_client:347]:  characteristic 0x2A01, handle 0x9, properties 0x2 			- Appearance
[01:13:17][I][ble_client:347]:  characteristic 0x2A04, handle 0xb, properties 0x2 			- Peripheral Preferred Connection Parameters
[01:13:17][I][ble_client:161]: Service UUID: 0x180A 									- Device Information service
[01:13:17][I][ble_client:162]:   start_handle: 0xc  end_handle: 0xe
[01:13:17][I][ble_client:347]:  characteristic 0x2A50, handle 0xe, properties 0x2 			- PnP ID
[01:13:17][I][ble_client:161]: Service UUID: 0x1812										- Human Interface Device service
[01:13:17][I][ble_client:162]:   start_handle: 0xf  end_handle: 0x1f
[01:13:17][I][ble_client:347]:  characteristic 0x2A4E, handle 0x11, properties 0x6 			- Protocol Mode
[01:13:17][I][ble_client:347]:  characteristic 0x2A4D, handle 0x13, properties 0x12 		- Report
[01:13:17][I][ble_client:347]:  characteristic 0x2A4D, handle 0x17, properties 0x12 		- Report
[01:13:17][I][ble_client:347]:  characteristic 0x2A4B, handle 0x1b, properties 0x2 			- Report Map
[01:13:17][I][ble_client:347]:  characteristic 0x2A4A, handle 0x1d, properties 0x2 			- HID Information
[01:13:17][I][ble_client:347]:  characteristic 0x2A4C, handle 0x1f, properties 0x4 			- HID Control Point
[01:13:17][I][ble_client:161]: Service UUID: 0x180F										- Battery service
[01:13:17][I][ble_client:162]:   start_handle: 0x20  end_handle: 0x23
[01:13:17][I][ble_client:347]:  characteristic 0x2A19, handle 0x22, properties 0x12   		- Battery Level
```
If there is anything else I can provide or test, then please let me know. 

Thank you very much!

yildiraymeric, 2023-07-11:
Hi,

I think what you are looking for is this project https://github.com/fsievers22/esphome-ble-remote

whc2001, 2023-11-29:
Just a note: these bluetooth shutter buttons turn themselves off after a certain period to save power. You need to press at least two times (first time to wake it up and make the connection, second to register the key press) to get it working after idling for a while, which I found quite annoying.