esphome/feature-requests #2024: ESP32-D0WD on Sonoff THR316

sirronb, 2022-12-20:
The Sonoff THR316 has an Espressif ESP32-D0WD V3 chip. Please add ESP Board support for this chip.

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-12-21:
Isn't that just a normal ESP32 flavour?

sirronb, 2022-12-22:
> Isn't that just a normal ESP32 flavour?

Hi @nagyrobi , I have tried that amongst other board types (esp32dev ; nodemcu-s) and they all fail to map the gpio pins correctly, or so it seems. I have also been following a lot of chat on discord. Seems I'm not alone.

isemenko, 2023-02-12:
@sirronb, it works fine for me using `esp32dev` board.

A couple of things that were not clear to me before I made it work:
1. esp32 is power hungry, so my device was not able to boot from usb serial after initial upload. Switched to AC and continued with OTA updates

2. pins are described here, not exactly TH16, but also r3/esp32, so matches
https://esphome.io/devices/sonoff.html#sonoff-thr316d 
Just in case, major ones:
```
GPIO0 - Button
GPIO21 - Relay
GPIO16 - red power led
GPIO15 - blue wifi led
GPIO13 - green led
```

sirronb, 2023-02-12:
Hi @isemenko,

I have tried many esp32 board options, esp32dev being one of them. I’m
aware of the power hungry issue and also realized that the board had to be
connected to AC.

I have also used the pin configurations as you suggested, but the board
still goes “available - unavailable - available - unavailable continuously.
Also pin GPIO25 is used for the dallas sensor which is not detected by
esphome.

For now I am using tasmota as a temporary work around until the problem can
be resolved. As said here, tasmota works perfectly out-of-the-box with the
ESP32-D0WD-V3.
I work like to stick to esphome with all my devices to keep consistency
throughout.

Please note that I do not have anything bad to say about tasmota. It is
brilliant, but one has to use MQTT with tasmota. Hence my preference is to
go with esphome throughout.

I’ll patiently await for the esphome dev-team to see if they can make it
work.

Warm regards
sirronb.

RoganDawes, 2023-03-15:
Sorry to hijack this thread, I am hoping to use a THR316 to interface with an NTC thermistor, rather than a DS18B20. Can anyone tell me if the PIN connected to the RJ9 interface is ADC-capable? Or if it is digital only?

RoganDawes, 2023-03-17:
To answer my own question: The GPIO25 pin is connected to ADC2, which is effectively unusable when WiFi is in use. While it *might* be possible to turn WiFi off, sample the pin, and then turn WiFi back on, I decided to go with an external ADC, presenting a DS18b20 interface to the THR320. A sample sketch can be found here: https://github.com/orgua/OneWireHub/blob/main/examples/DS18B20_thermometer/DS18B20_thermometer.ino
This can be implemented on a smaller Arduino-capable chip, like a 328, or (possibly) an ATTiny412 which I will be attempting. This will let me sample my 10k NTC thermistor, convert that into a digital reading, poll that using 1-Wire, and ultimately report that value.

closed 2023-07-19 by sirronb