esphome/feature-requests #2025: Make log optional in web_server

stevetz, 2022-12-21:
I’d like to present family members with a simple webui - where they don’t need to see the logs.

Similarly adding an optional variable to the implemented `ota:` one would be really useful, to me at least.

I’m afraid I have no understanding of how much work this would take to add, so please forgive me if it’s ridiculous!

ssieb, 2022-12-21:
I don't understand what you're suggesting about the `ota:`.  Can you clarify?

stevetz, 2022-12-21:
My apologies, I meant that just to compare a new potentially optional setting for the log might be implemented similarly to the optional ota -
```
web_server:
  ...
  ota: false
  # the feature request would be this
  log: false
```

closed 2023-06-21 by jesserockz