esphome/feature-requests #2028: Add sensors support for remote_receiver

CmPi, 2022-12-21:
**Describe the problem you have/What new integration you would like**

I managed to implement a new protocol (TX3-TX4, WS7000 weather sensors from Lacrosse) for the component remote_base and I would like now to declare sensors to report temperature, humidity etc... Unfortunately, only binary sensors are supported (IR or 433MHz remote controllers used to switch on/ switch off). 

**Please describe your use case for this integration and alternatives you've tried:**

I tried to create a custom sensor but I did not find how to communicate between the remote_receiver component and the custom sensor.

**Additional context**

I tried to modify deeply __init__.py but it looks to not be enough....

nagyrobi, 2022-12-21:
Please join the dev channel on our Discord server with this request.

CmPi, 2022-12-22:
Thanks. It's done:
https://discord.com/channels/429907082951524364/1055361792420884540