esphome/feature-requests #2030: Support for Shelly Plus H&T

retrry, 2022-12-22:
**Describe the problem you have/What new integration you would like**
I would like to have support for Shelly Plus H&T device. It is based on ESP32 processor.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to use these devices for temperature monitoring in my home with Home Assistant.

**Additional context**
https://www.shelly.cloud/en/products/shop/shelly-plus-h-and-t
https://kb.shelly.cloud/knowledge-base/shelly-plus-h-t
I can buy couple devices for a person, who will implement support for it in ESPHome.

avilleret, 2023-03-10:
I guess the main issue here is the E-paper graphic display that needs to be retro engineered to understand how it is wired and how to control it with custom firmware.
The rest should be an ESP32 platform with hygrometry and temperature sensors which should be not so difficult to interface with ESPHome.

Robbe-B, 2023-11-11:
I'm really willing to give up the display of that device in favour of having faster sensor updates from the custom firmware.

I was planning on using this device to control humidity inside my bathroom by triggering the bathroom fan. (e.g. during a rapid humidity increase when taking a shower)

But unfortunately this device (out of the box) is just too slow for this. The fastest updates you get (when powering via usb) is each 5 minutes.

I'm really hoping that this can change when we flash this device with esphome. 
Even is the display is nog working anymore, I would be happy.

grischard, 2024-06-05:
Shelly [claims](https://www.reddit.com/r/ShellyUSA/comments/1aofmrd/anyone_having_trouble_with_shelly_ht_gen3_staying/kqf4mhz/) that this is necessary so that the radio doesn't heat up the thermometer.

Robbe-B, 2024-06-05:
> Shelly [claims](https://www.reddit.com/r/ShellyUSA/comments/1aofmrd/anyone_having_trouble_with_shelly_ht_gen3_staying/kqf4mhz/) that this is necessary so that the radio doesn't heat up the thermometer

@grischard , I'm aware of this, would this also be the case for the newer Gen3 models ?

grischard, 2024-06-05:
@Robbe-B yes. While esphome would be cool on these (the e-ink display looks nice), they probably only nerf the thermometer and updates because they have no other choice. Overall, it feels like a poorly designed device that's full of quirks.

Robbe-B, 2024-06-05:
> @Robbe-B yes. While esphome would be cool on these (the e-ink display looks nice), they probably only nerf the thermometer and updates because they have no other choice. Overall, it feels like a poorly designed device that's full of quirks.

@grischard, 
I totally agree, 
I was kinda disappointed when I discovered all this after buying it.
I even bought the Gen3 because they promised faster updates. 
But really, nothing changed compared to the Gen2 models.