esphome/feature-requests #2031: Integration sensor reset to specific value

mrkvo, 2022-12-22:
**Describe the problem you have/What new integration you would like**

Hi! I would like to ask you about modification of integration sensor - reset is now possible only to zero. 
Also, cheat over lambda does not work, the integration value is in protected space.

**Please describe your use case for this integration and alternatives you've tried:**

Reset action with parameter (float)

**Additional context**

None

Karatzie, 2023-11-12:
I would also like to use this feature.

BR.
Karatzie

mrkvo, 2024-02-16:
Please, any progress?

olicooper, 2025-07-17:
I needed the same functionality so I came up with a really fragile/hacky way to do it that takes advantage of C++ inheritance and pointers (so it could break if an esphome update changes the class definition etc.)

This example updates the integration sensor with the id `integration_sensor`.

```yaml
number:
  - platform: template
    name: Set Integration Value
    entity_category: config
    mode: box
    max_value: 100
    min_value: 0
    update_interval: never
    step: .1
    restore_value: false
    set_action:
      then:
        lambda: |-
          class IntegrationAccessHack : public integration::IntegrationSensor {
            public:
              static void publish_and_save(integration::IntegrationSensor& base, double value) {
                  float integration::IntegrationSensor::*p_last_value = &IntegrationAccessHack::last_value_;
                  base.*p_last_value = value;
                  void (integration::IntegrationSensor::*p_func)(double v) = &IntegrationAccessHack::publish_and_save_;
                  (base.*p_func)(value);
              }
          };
          // Update the Integration sensor value
          IntegrationAccessHack::publish_and_save(*id(integration_sensor), x);
```