esphome/feature-requests #2034: Add support for Shelly Plus Addon

raidr1, 2022-12-24:
**Describe the problem you have/What new integration you would like**

Add support for Shelly Plus Addon

**Please describe your use case for this integration and alternatives you've tried:**

it would be nice to integrate with all Shelly Plus devices

**Additional context**

nagyrobi, 2022-12-24:
The addon is just a galvanic isolated adapter for GPIO pins. It's transparent to ESPHome.

DHowett, 2023-01-01:
I don’t believe it’s a simple passthrough. It connects to the serial/debug header on the Plus devices and expands the host device with support for 1-Wire, one additional ADC and one additional GPIO. Those aren’t exposed via that header.

In addition, Shelly documents the addon as using a …

> Shelly proprietary serial interface to connect to a Shelly Plus device
> _[source](https://kb.shelly.cloud/knowledge-base/shelly-plus-add-on)_

jtricerolph, 2023-01-07:
I opened mine up and the pins trace through to seperate pins on the isolator chip, I'm pretty sure it's not using uart based on the chip model number.

jtricerolph, 2023-01-13:
> Thanks for tracing these out! That makes sense. Yeah, RX0/TX0 can be re-muxed to be I/O, and then we have IO19 and IO0 on the pin header. Good call.
> 
> I'm interested in the extra analog input. None of those four pins seem (according to the datasheet; I don't know if they do in practice) to support ADC... 🤷🏻 

Don't quote me on the above tracing but the following 😅
But one I put that I think is the analoge trace goes through a seperate chip labeled 3PEAK 393a BQLh and although can't find an exact match on Google maybe it's doing some pre processing for that?

jowgn, 2023-01-17:
[image]
[image]
[image]
[image]

OldSilvaRanga, 2023-04-07:
Any movement on the use of the Addon within ESPHome?  Would be a great addition to the existing shelly gear.

pdw-mb, 2023-04-11:
> But one I put that I think is the analoge trace goes through a seperate chip labeled 3PEAK 393a BQLh and although can't find an exact match on Google maybe it's doing some pre processing for that?

It's this one - [LM393A datasheet](https://datasheet.lcsc.com/lcsc/2203101903_3PEAK-LM393A-SR_C2690180.pdf)

andypnz, 2023-04-27:
Is there any possibility of using the addon with ESPHome? Thanks you!

pdw-mb, 2023-04-30:
I've spent some time figuring out how the ADC works on this.  It actually appears as the duty cycle on the RX pin (GPIO3).  This config appears to read it correctly:

```
sensor:
  - platform: duty_cycle
    pin: 3
    name: Duty Cycle Sensor
    update_interval: 5s
```

The board is generating a sawtooth input to one pin of the comparator, with the analog input on the other, giving a digital output with duty cycle representing the analog input.  I don't fully understand the circuit, as it appears that the frequency and amplitude of the signal vary according to the resistors.

I think the other pins are: GPIO19 - digital input, GPIO0 1-wire TX, GPIO1 1-wire RX.  I've not tried, but I think that  https://github.com/esphome/esphome/pull/1346 would allow you to use the DS18B20 sensors with it, if you set `pin: 1`, `pin_a: 0`.  It looks like the PR needs some work to sort out some compilation issues.

ashp8i, 2023-05-24:
I have the dht working [https://github.com/ashp8i/esphome-configs/tree/main/my_external_components/test4/shelly_dht](https://github.com/ashp8i/esphome-configs/tree/main/my_external_components/test4/shelly_dht)

using the following config: -
```
sensor:
  - platform: shelly_dht
    pin: GPIO1
    pin_a: GPIO0
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 60s
```

I need to find a way how to leave the original component alone but add ifdef or new methods for the shelly that uses two pins, same goes for the dallas, my plan for the dallas side is to add support for 1-Wire via RMT peripheral (ESP32's) and make the bitbanging more robust for the ESP8266, and support Switch and I2C bridge, sorry not high-jacking thread, trying to give this some direction.

amzaldua, 2023-11-10:
Is it possible to integrate a CT sensor with the Shelly 1 Plus using the add-on module? Currently, my Shelly Plus 1PM, connected to my boiler, is experiencing issues with overheating. Could using a Shelly Plus 1 with a dry contact potentially resolve this problem?

yasapl, 2023-12-20:
> I've spent some time figuring out how the ADC works on this. It actually appears as the duty cycle on the RX pin (GPIO3). This config appears to read it correctly:
> 
> ```
> sensor:
>   - platform: duty_cycle
>     pin: 3
>     name: Duty Cycle Sensor
>     update_interval: 5s
> ```
> 
> The board is generating a sawtooth input to one pin of the comparator, with the analog input on the other, giving a digital output with duty cycle representing the analog input. I don't fully understand the circuit, as it appears that the frequency and amplitude of the signal vary according to the resistors.
> 
> I think the other pins are: GPIO19 - digital input, GPIO0 1-wire TX, GPIO1 1-wire RX. I've not tried, but I think that [esphome/esphome#1346](https://github.com/esphome/esphome/pull/1346) would allow you to use the DS18B20 sensors with it, if you set `pin: 1`, `pin_a: 0`. It looks like the PR needs some work to sort out some compilation issues.

Hi, would you be able to suggest a configuration for an NTC resistor to use ut as a temperature sensur (underfloor heating, 10Kohm at 25deg C)? Seems that the original shelly fw does support it so the hardware should be OK.

Thank you in advance!
Yasa

pdw-mb, 2023-12-20:
> Hi, would you be able to suggest a configuration for an NTC resistor to use it as a temperature sensor (underfloor heating, 10Kohm at 25deg C)? Seems that the original shelly fw does support it so the hardware should be OK. I know I could use dallas sensor but the ntc is already buried in the floor.

This is actually exactly what I use my Shelly add-ons for :smiley: 

```yaml
  - platform: ntc
    sensor: resistance_sensor
    calibration:
      - 11.4kOhm -> 26.6°C
      - 13.17kOhm -> 24°C
      - 16.17kOhm -> 18.8°C
    name: Floor Temperature
    id: floor_temperature

  # Example source sensors:
  - platform: resistance
    id: resistance_sensor
    sensor: add_on_adc
    configuration: DOWNSTREAM
    resistor: 10kOhm
    name: Resistance Sensor

  - platform: duty_cycle
    pin: 3
    id: add_on_adc
    name: "Add-on ADC"
    update_interval: 5s
    unit_of_measurement: "V"
    filters:
      - offset: -2.0
      - multiply: 0.033
```

The calibration was based on measuring the existing NTC and taking readings from the existing floor controller.

I can't remember how I arrived at the `offset` value for the duty cycle filter.  The `0.033` is to transfer 100% duty to 3.3V.

yasapl, 2023-12-21:
Thank you @pdw-mb. I will definitely try in once my add-on arrives.

larsskogshusabf, 2023-12-25:
Me too. My Addon arrives on Friday.

KaufDirEins, 2024-03-12:
Any news on using dallas ? My addon arrived today. And i cant get it to work

JKoss2, 2024-05-23:
Does anyone think it would be possible to run a PWM signal on GPIO19 of the add-on? Perhaps in combination with a DHT22 (since that uses GPIO1 according to [this comment](https://github.com/arendst/Tasmota/discussions/18885#discussioncomment-7295482) )? The PWM signal would be to adjust the speed of a fan. I know its possible to control a PWM fan in ESPHome, but I am unsure if the galvanic isolation of the add-on would affect the signal.

m1k3f15h, 2024-07-16:
Hy @KaufDirEins 

> Any news on using dallas ? My addon arrived today. And i cant get it to work

Did you already get them to work?

Mike

ShifuSonny, 2024-11-08:
any new info?

Sonny

m1k3f15h, 2024-11-12:
hy @ShifuSonny 
I Use this: https://github.com/persuader72/esphome-components
and it works perfect!

klim123123, 2024-11-12:
> hy @ShifuSonny
> I Use this: https://github.com/persuader72/esphome-components
> and it works perfect!

Temperature Sensor AddOn for Shelly 1/1PM and Shelly Plus Addon are not the same.

m1k3f15h, 2024-11-13:
> > hy @ShifuSonny
> > I Use this: https://github.com/persuader72/esphome-components
> > and it works perfect!
> 
> Temperature Sensor AddOn for Shelly 1/1PM and Shelly Plus Addon are not the same.

Hy
Oh.. I didn't see this :O

but i use the component with the shelly plus addon!

mike

klim123123, 2024-11-13:
> but i use the component with the shelly plus addon!

please provide your configuration

pdw-mb, 2024-11-13:
> Does anyone think it would be possible to run a PWM signal on GPIO19 of the add-on? 
> 
> Edit (11Jul2024): After testing that the PWM signal correctly controls a fan via the 7 pin female header on the back of a Shelly Plus 1PM, I've concluded that the galvanic isolation in the Shelly Plus Add-On prevents the PWM signal from passing through to the add-on terminals.

Late reply to this one, but yes, that is the case. The opto-isolator only passes each pin in one direction, so GPIO19 can only be used as an input.  That's also why the "1-wire" connection is broken out into separate TX and RX pins.

m1k3f15h, 2024-11-18:
Did anyone get the Digital Input on the Shelly Plus Addon to work? (Shelly2PM)

ashp8i, 2025-12-19:
> aizerin left a comment [(esphome/feature-requests#2034)](https://github.com/esphome/feature-requests/issues/2034#issuecomment-3676305284)
>
> I made addon works with Shelly 1PM Gen3. DHT + NTC. I used configuration from posts above and just changed pins to match new shelly device.
>
> Full config reference:
>
> substitutions:
>   friendly_name: thermo_obyvak
>   floor_max_temp: "37.0"
>   floor_cooldown_temp: "35.0"
>
> esphome:
>   name: thermo-one
>   friendly_name: $friendly_name
>
> esp32:
>   board: esp32-c3-devkitm-1
>   framework:
>     type: esp-idf
>
> external_components:
>   - source:
>       type: git
>       url: https://github.com/ashp8i/esphome-configs
>       path: my_external_components/test4
> [... 252 lines cut ...]
>       - switch.turn_off: relay
>     visual:
>       min_temperature: 10 °C
>       max_temperature: 25 °C
>       temperature_step: 1 °C

please fork my code, I have not worked on that piece for ages

aizerin, 2025-12-19:
> please fork my code, I have not worked on that piece for ages
> […](#)

I wanted to, but it still working :)

majoto85, 2026-02-02:
I wrote a little bit of code for the Shelly Plus Add On (or more accurately, had it written :-) ).
DS18B20, Analog In, and Digital In work with it. I didn't have a DHT22 lying around, so this component is missing.

[https://github.com/majoto85/esphome-shelly-plus-addon](https://github.com/majoto85/esphome-shelly-plus-addon)

Disclaimer:
Vibecodes with Claude, figuring out the timings became too annoying for me after a while.