esphome/feature-requests #2035: Support for ens160 air quality sensor

MrSuicideParrot, 2022-12-24:
**Describe the problem you have/What new integration you would like**
Add support for the ens160 air quality sensor.
**Please describe your use case for this integration and alternatives you've tried:**

A new type of air quality sensor has started to appear in the electronic shops, the ens160. It supports VOC, CO2 and AQI measurements and esphome doesn't support it.

I'm working on a pull request to add support for this sensor. The goal of this issue is to track any progress. 

**Additional context**

https://www.sciosense.com/products/environmental-sensors/digital-multi-gas-sensor/
https://learn.adafruit.com/adafruit-ens160-mox-gas-sensor
https://learn.sparkfun.com/tutorials/sparkfun-indoor-air-quality-sensor---ens160-qwiic-hookup-guide

MrSuicideParrot, 2023-01-01:
I was testing my implementation before doing the pull request and noticed that there is already a pull request https://github.com/esphome/esphome/pull/4243.

vincentscode, 2023-01-02:
Hey, sorry @MrSuicideParrot I should've mentioned this issue earlier to prevent duplicate work. :)
Feel free to look over my implementation and give any feedback you might have!

phzi, 2023-05-31:
ACK; Please merge

Tntdruid, 2023-10-02:
Still not merged?

michaelsidenius, 2023-10-22:
I would appriciate if this pull request could be merged. I would gladly test it, have a bunch of ENS160+AHT2X boards laying around.

vincentscode, 2023-10-22:
@michaelsidenius You can already test components from PRs using `external_component`:
```yaml
external_components:
  - source: github://pr#4243
    components:
      - ens160
```
No clue why the request has not yet been merged tho.

Epimedium, 2023-11-03:
Hi,
I was also looking for ENS160 support for esphome. I've used the solution above and al works fine except no data is being read from the sensor. Can you give me a hint? Maybe my sensor is broken.

Best regards,
Erik

This is my code:
```
external_components:
  - source: github://pr#4243
    components:
      - ens160

i2c:
  sda: D1
  scl: D2

sensor:
  - platform: ens160
    update_interval: 10s  
    eco2:
      name: "eCO2 Value"
    tvoc:
      name: "Total Volatile Organic Compound"
    aqi:
      name: "Air Quality Index"
```

These are the logs:
[08:18:29][C][logger:416]: Logger:
[08:18:29][C][logger:417]:   Level: DEBUG
[08:18:29][C][logger:418]:   Log Baud Rate: 115200
[08:18:29][C][logger:420]:   Hardware UART: UART0
[08:18:29][C][i2c.arduino:053]: I2C Bus:
[08:18:29][C][i2c.arduino:054]:   SDA Pin: GPIO5
[08:18:29][C][i2c.arduino:055]:   SCL Pin: GPIO4
[08:18:29][C][i2c.arduino:056]:   Frequency: 50000 Hz
[08:18:29][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[08:18:29][I][i2c.arduino:069]: Results from i2c bus scan:
[08:18:29][I][i2c.arduino:075]: Found i2c device at address 0x38
[08:18:29][I][i2c.arduino:075]: Found i2c device at address 0x53
[08:18:29][C][ens160:278]: ENS160:
[08:18:29][D][ens160:300]: Setup successful
[08:18:29][I][ens160:303]: Firmware Version: 5.4.6
[08:18:29][C][ens160:306]:   Address: 0x53
[08:18:29][C][ens160:307]:   Update Interval: 10.0s
[08:18:29][C][ens160:308]:   CO2 Sensor: 'eCO2 Value'
[08:18:29][C][ens160:308]:     Device Class: 'carbon_dioxide'
[08:18:29][C][ens160:308]:     State Class: 'measurement'
[08:18:29][C][ens160:308]:     Unit of Measurement: 'ppm'
[08:18:29][C][ens160:308]:     Accuracy Decimals: 0
[08:18:29][C][ens160:308]:     Icon: 'mdi:molecule-co2'
[08:18:29][C][ens160:309]:   TVOC Sensor: 'Total Volatile Organic Compound'
[08:18:29][C][ens160:309]:     Device Class: 'volatile_organic_compounds'
[08:18:29][C][ens160:309]:     State Class: 'measurement'
[08:18:29][C][ens160:309]:     Unit of Measurement: 'ppb'
[08:18:29][C][ens160:309]:     Accuracy Decimals: 0
[08:18:29][C][ens160:309]:     Icon: 'mdi:radiator'
[08:18:29][C][ens160:310]:   AQI Sensor: 'Air Quality Index'
[08:18:29][C][ens160:310]:     Device Class: 'aqi'
[08:18:29][C][ens160:310]:     State Class: 'measurement'
[08:18:29][C][ens160:310]:     Unit of Measurement: 'index'
[08:18:29][C][ens160:310]:     Accuracy Decimals: 0
[08:18:29][C][ens160:310]:     Icon: 'mdi:chemical-weapon'
[08:18:29][C][ens160:316]:   Compensation: Not configured
[08:18:29][C][captive_portal:088]: Captive Portal:
[08:18:29][C][mdns:115]: mDNS:
[08:18:29][C][mdns:116]:   Hostname: co2-meter
[08:18:29][C][ota:097]: Over-The-Air Updates:
[08:18:29][C][ota:098]:   Address: co2-meter.local:8266
[08:18:29][C][ota:101]:   Using Password.
[08:18:29][C][api:139]: API Server:
[08:18:29][C][api:140]:   Address: co2-meter.local:6053
[08:18:29][C][api:142]:   Using noise encryption: YES
[08:18:31][D][api.connection:1050]: Home Assistant 2023.10.5 (192.168.50.102): Connected successfully
[08:18:33][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:18:43][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:18:53][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:19:03][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:19:13][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:19:23][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[08:19:33][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready

Koen00, 2023-11-08:
> Hi, I was also looking for ENS160 support for esphome. I've used the solution above and al works fine except no data is being read from the sensor. Can you give me a hint? Maybe my sensor is broken.
> 
> Best regards, Erik
> 

I have used the same code but with 30s update interval. maybe worth a try?

> @michaelsidenius You can already test components from PRs using `external_component`:
> 
> ```yaml
> external_components:
>   - source: github://pr#4243
>     components:
>       - ens160
> ```
> 
> No clue why the request has not yet been merged tho.

Works like a charm. Got a ENS160 + AHT2x sensor on a ESP32 feather working. Thx a lot!

Tntdruid, 2023-11-08:
@Epimedium Did it run for 1 h?

Fist boot need 1 h before and data.

Koen00, 2023-11-08:
> @Epimedium Did it run for 1 h?
> 
> Fist boot need 1 h before any data.

Yes, That's correct. But it's a shame that it will also take an hour after a software reboot without any power disconnection. (2nd power on)

Isn't it an option to poll the current time and see when it was last "online" so it did not cooldown. (I think that's why it needs 1 hour)? 

`<html>
<body>

15:17:38 | [D] | [text_sensor:064] | 'esp32-03 Battery status': Sending state 'In use'
-- | -- | -- | --
15:17:57 | [D] | [sensor:094] | 'esp32-03 eCO2 Value': Sending state 605.00000 ppm with 0 decimals of accuracy
15:17:57 | [D] | [sensor:094] | 'esp32-03 Total Volatile Organic Compound': Sending state 135.00000 ppb with 0 decimals of accuracy
15:17:57 | [D] | [sensor:094] | 'esp32-03 Air Quality Index': Sending state 2.00000 index with 0 decimals of accuracy
15:18:01 | [D] | [switch:012] | 'esp32-03 Restart' Turning ON.
15:18:01 | [D] | [switch:055] | 'esp32-03 Restart': Sending state OFF
15:18:01 | [I] | [restart:016] | Restarting device...
15:18:01 | [I] | [app:134] | Rebooting safely...

</body>
</html>`

software reboot:

`<html>
<body>

15:18:18 | [I] | [ens160:179] | ENS160 readings unavailable - 1 hour startup required after first power on
-- | -- | -- | --
15:18:42 | [D] | [aht10:083] | AHT10 is busy, waiting...
15:18:42 | [D] | [sensor:094] | 'esp32-03 AHT Temperature': Sending state 16.92581 °C with 2 decimals of accuracy
15:18:42 | [D] | [sensor:094] | 'esp32-03 AHT Humidity': Sending state 49.95728 % with 2 decimals of accuracy

</body>
</html>`

Koen00, 2023-11-08:
Once booting and waiting on the sensor, I would like to see when it is ready to get data. Like a boolean to poll/use in a Lambda function.  Now I use the value NaN to determine if there is a value or no value (no value is booting) but could also have other reasons to show NaN, like a malfunction or disconnection. Would it be easy to expose this information? Or is it already available?

`      if (isnan(id(iaq_index).state)) {
        return {"Sensor not yet ready"};
      }`

Koen00, 2023-11-08:
I am now using it more and investigating (trying) the source code and I have some questions.

1. I would like to see the state of the sensor. I noticed there are booleans and an enumerator giving the state of the sensor.    NORMAL_OPERATION, WARMING_UP,  INITIAL_STARTUP,  INVALID_OUTPUT. Is there a way to get this out of the sensor? To use it in a lambda function?
2. Temperature and Humidity correction is also in the source code. But trying to configure this with startup (like sensor CCS811) is not successful. 

```
sensor:
  - platform: ens160
    id: ens160Sensor
    update_interval: never
    eco2:
      name: "$node_name eCO2 Value"
      id: eco2
    tvoc:
      name: "$node_name Total Volatile Organic Compound"
      id: tvoc
    aqi:
      name: "$node_name Air Quality Index"
    temperature: AHTTemp
    humidity: humi
```

3. How is the air quality index calculated and what does it say? Is it like on [iaquk.org.uk](http://www.iaquk.org.uk/ )? I use the code below to define it in human words how the quality is, used from this [post](https://community.home-assistant.io/t/example-indoor-air-quality-text-sensor-using-ccs811-sensor/125854).  Maybe I'm just reading things wrong. 

```
text_sensor:  
  - platform: template
    name: "Livingroom IAQ"
    icon: "mdi:air-filter"
    lambda: |-
      id(iaq_index) = 0;
      
      /*
       * Transform indoor humidity values to IAQ points according to Indoor Air Quality UK: 
       * http://www.iaquk.org.uk/
       */
      if (id(humi).state < 10 or id(humi).state > 90) {
        id(iaq_index) += 1;
      }
      else if (id(humi).state < 20 or id(humi).state > 80) {
        id(iaq_index) += 2;
      }
      else if (id(humi).state < 30 or id(humi).state > 70) {
        id(iaq_index) += 3;
      }
      else if (id(humi).state < 40 or id(humi).state > 60) {
        id(iaq_index) += 4;
      }
      else if (id(humi).state >= 40 and id(humi).state <= 60) {
        id(iaq_index) += 5;
      }
      
      /*
       * Transform eCO2 values to IAQ points according to Indoor Air Quality UK: 
       * http://www.iaquk.org.uk/
       */
      if (id(eco2).state <= 600) {
        id(iaq_index) += 5;
      }
      else if (id(eco2).state <= 800) {
        id(iaq_index) += 4;
      }
      else if (id(eco2).state <= 1500) {
        id(iaq_index) += 3;
      }
      else if (id(eco2).state <= 1800) {
        id(iaq_index) += 2;
      }
      else if (id(eco2).state > 1800) {
        id(iaq_index) += 1;
      }
      
      /*
       * Transform TVOC values to IAQ points according to German environmental guidelines: 
       * https://www.repcomsrl.com/wp-content/uploads/2017/06/Environmental_Sensing_VOC_Product_Brochure_EN.pdf
       */
      if (id(tvoc).state <= 65) {
        id(iaq_index) += 5;
      }
      else if (id(tvoc).state <= 220) {
        id(iaq_index) += 4;
      }
      else if (id(tvoc).state <= 660) {
        id(iaq_index) += 3;
      }
      else if (id(tvoc).state <= 2200) {
        id(iaq_index) += 2;
      }
      else if (id(tvoc).state > 2200) {
        id(iaq_index) += 1;
      }

      /*
       * Transform IAQ index to human readable text according to Indoor Air Quality UK: 
       * http://www.iaquk.org.uk/
       */
      ESP_LOGD("main", "Current IAQ index %d", id(iaq_index));
      
      if (id(iaq_index) <= 6) {
        return {"Unhealty"};
      }
      else if (id(iaq_index) <= 9) {
        return {"Poor"};
      }
      else if (id(iaq_index) <= 12) {
        return {"Moderate"};
      }
      else if (id(iaq_index) <= 14) {
        return {"Good"};
      }
      else if (id(iaq_index) > 14) {
        return {"Excellent"};
      }
      
      return {};
    update_interval: 60s
```

Epimedium, 2023-11-09:
> @Epimedium Did it run for 1 h?
> 
> Fist boot need 1 h before any data.

Hi Tntdruid, Koen,

Thanks for your reply. I was aware of the startup time and even after 6 hours there was no data. 
I'm trying now with a refresh of 30 seconds but I doubt thats going to bring any changes.

Best, Erik

Epimedium, 2023-11-09:
I just tested with 30 refresh. A slight change: right after startup some readings were done.

[09:00:34][I][i2c.arduino:069]: Results from i2c bus scan:
[09:00:34][I][i2c.arduino:075]: Found i2c device at address 0x38
[09:00:34][I][i2c.arduino:075]: Found i2c device at address 0x53
[09:00:34][C][ens160:278]: ENS160:
[09:00:34][D][ens160:300]: Setup successful
[09:00:34][I][ens160:303]: Firmware Version: 5.4.6
[09:00:34][C][ens160:306]:   Address: 0x53
[09:00:34][C][ens160:307]:   Update Interval: 30.0s
[09:00:34][C][ens160:308]:   CO2 Sensor: 'eCO2 Value'
[09:00:34][C][ens160:308]:     Device Class: 'carbon_dioxide'
[09:00:34][C][ens160:308]:     State Class: 'measurement'
[09:00:34][C][ens160:308]:     Unit of Measurement: 'ppm'
[09:00:34][C][ens160:308]:     Accuracy Decimals: 0
[09:00:34][C][ens160:308]:     Icon: 'mdi:molecule-co2'
[09:00:34][C][ens160:309]:   TVOC Sensor: 'Total Volatile Organic Compound'
[09:00:34][C][ens160:309]:     Device Class: 'volatile_organic_compounds'
[09:00:34][C][ens160:309]:     State Class: 'measurement'
[09:00:34][C][ens160:309]:     Unit of Measurement: 'ppb'
[09:00:34][C][ens160:309]:     Accuracy Decimals: 0
[09:00:34][C][ens160:309]:     Icon: 'mdi:radiator'
[09:00:34][C][ens160:310]:   AQI Sensor: 'Air Quality Index'
[09:00:34][C][ens160:310]:     Device Class: 'aqi'
[09:00:34][C][ens160:310]:     State Class: 'measurement'
[09:00:34][C][ens160:310]:     Unit of Measurement: 'index'
[09:00:34][C][ens160:310]:     Accuracy Decimals: 0
[09:00:34][C][ens160:310]:     Icon: 'mdi:chemical-weapon'
[09:00:34][C][ens160:316]:   Compensation: Not configured
[09:00:34][C][captive_portal:088]: Captive Portal:
[09:00:34][C][mdns:115]: mDNS:
[09:00:34][C][mdns:116]:   Hostname: co2-meter
[09:00:34][C][ota:097]: Over-The-Air Updates:
[09:00:34][C][ota:098]:   Address: co2-meter.local:8266
[09:00:34][C][ota:101]:   Using Password.
[09:00:34][C][api:139]: API Server:
[09:00:34][C][api:140]:   Address: co2-meter.local:6053
[09:00:34][C][api:142]:   Using noise encryption: YES
[09:00:36][D][api:102]: Accepted ip
[09:00:36][W][component:214]: Component api took a long time for an operation (0.29 s).
[09:00:36][W][component:215]: Components should block for at most 20-30ms.
[09:00:37][D][api.connection:1050]: Home Assistant (): Connected successfully
[09:00:47][D][sensor:093]: 'eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[09:00:47][D][sensor:093]: 'Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
[09:00:47][D][sensor:093]: 'Air Quality Index': Sending state 0.00000 index with 0 decimals of accuracy
[09:01:17][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[09:01:47][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
~
[09:05:17][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[09:05:22][I][ota:117]: Boot seems successful, resetting boot loop counter.
[09:05:47][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
~
[10:30:47][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[10:31:17][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready

xxddff, 2023-11-14:
I have the same problem as @Epimedium. It is recommended to turn on the VERBOSE log level to view more detailed information.
On another board (ens160+aht21), ens160 runs normally, but the current one being tested (ens160 alone) reports 0 data at the beginning and has the following bits. A normally running ens160 should not be like this.
```
[08:21:14][V][ens160:157]: Status: ENS160 STATAS bit    0x0
[08:21:14][V][ens160:159]: Status: ENS160 STATER bit    0x0
[08:21:14][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[08:21:14][V][ens160:162]: Status: ENS160 NEWDAT bit    0x0
[08:21:14][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
```

Koen00, 2023-11-19:
Hi, 

I have run the code a lot longer now and have seen some improvements. Spent some time extra and noticed some changes I would like to share and come back to my previous questions/statements. 

> 2. Temperature and Humidity correction is also in the source code. But trying to configure this with startup (like sensor CCS811) is not successful.

I have it working now using "compensation" as a statement as follows:

```  
sensor:
  - platform: ens160
    id: ens160Sensor
    update_interval: 60s
    eco2:
      name: "$node_name eCO2 Value"
      id: eco2
    tvoc:
      name: "$node_name Total Volatile Organic Compound"
      id: tvoc
    aqi:
      name: "$node_name Air Quality Index"
    compensation:
      temperature: AHTTemp
      humidity: humi
```

--------older post ---- 
> Yes, That's correct. But it's a shame that it will also take an hour after a software reboot without any power disconnection. (2nd power on)
> 
> Isn't it an option to poll the current time and see when it was last "online" so it did not cooldown. (I think that's why it needs 1 hour)?

During a long time of running it now recognizes that it is not anymore the initial boot. Now it needs like 3 minutes to warm up. I'm not sure if in the mean time an update in software has been taken place. Or my ESP32 is just a bit weird. It looks like it is working a lot better now: See log->

```
[18:33:20][I][ens160:185]: ENS160 readings not available yet - Warming up requires 3 minutes
[18:33:20][W][component:214]: Component interval took a long time for an operation (0.09 s).
[18:33:20][W][component:215]: Components should block for at most 20-30ms.
[18:33:29][D][esp32.preferences:114]: Saving 1 preferences to flash...
[18:33:29][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:33:32][D][sensor:094]: 'esp32-03 Battery Percentage': Sending state -257.69559 % with 1 decimals of accuracy
[18:34:20][D][aht10:084]: AHT10 is busy, waiting...
[18:34:20][D][sensor:094]: 'esp32-03 AHT Temperature': Sending state 24.91550 °C with 2 decimals of accuracy
[18:34:20][D][sensor:094]: 'esp32-03 AHT Humidity': Sending state 57.88527 % with 2 decimals of accuracy
```

closed 2023-11-24 by kbx81

elemakil, 2023-12-03:
Hey,

I have a combined AHT21+ENS160 sensor like [this package](https://www.androegg.de/shop/ens160aht21-co2-kohlendioxid-raumluft-vocs-tvoc-aql-sensor/) and I can observe similar behaviour or possibly the same as @Narqulie.

Initially, the sensor is in startup mode and I get the following readout:
````
[10:59:20][V][ens160:157]: Status: ENS160 STATAS bit    0x1
[10:59:20][V][ens160:159]: Status: ENS160 STATER bit    0x0
[10:59:20][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x02
[10:59:20][V][ens160:162]: Status: ENS160 NEWDAT bit    0x1
[10:59:20][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
````
After about one hour, the output shifts to 
````
[11:02:20][V][ens160:157]: Status: ENS160 STATAS bit    0x1
[11:02:20][V][ens160:159]: Status: ENS160 STATER bit    0x0
[11:02:20][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[11:02:20][V][ens160:162]: Status: ENS160 NEWDAT bit    0x1
[11:02:20][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
````
and I receive data:
````
[11:02:20][D][sensor:094]: 'eCO2': Sending state 466.00000 ppm with 0 decimals of accuracy
[11:02:20][D][sensor:094]: 'Total Volatile Organic Compounds': Sending state 55.00000 ppb with 0 decimals of accuracy
[11:02:20][D][sensor:094]: 'Air Quality Index': Sending state 1.00000 index with 0 decimals of accuracy
[11:02:20][V][ens160:247]: External temperature compensation: 28.9°C
[11:02:20][V][ens160:258]: External humidity compensation:    31.2%
````
Then, for some period the sensor sends correct data (shortest period: 15 minutes, longest period: about 12 hours) but eventually the output shifts to
````
[17:55:36][V][ens160:157]: Status: ENS160 STATAS bit    0x0
[17:55:36][V][ens160:159]: Status: ENS160 STATER bit    0x0
[17:55:36][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[17:55:36][V][ens160:162]: Status: ENS160 NEWDAT bit    0x0
[17:55:36][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
[17:55:36][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
````
At this point, the data that gets ingested by HomeAssistant is all-zeroes for eCO2, TVOC and AQI. Rebooting the ESP or removing and reconnecting power usually does not reset the sensor into its initial configuration. I have managed to reset it into its initial configuration - which causes it to go through the above steps once again - by resoldering the connections to the sensor.

I've checked the data in HomeAssistant and made some interesting observations. The following two screenshots show the temperature (from the AHT21) and TVOC (from the ENS160) over the same period of time. The gaps in the temperature profile are caused by me trying to reset the ENS160 sensor by removing it from power.

[image]
[image]

As you can see, there is a strong correlation between the temperature and the validity of the ENS160 data. It seems that the heater of the ENS160 significantly warms up the AHT21 as there is an increase of about 6 °C when the ENS160 is producing data. This is a bad sign for anyone wanting to use this sensor for environment monitoring. Furthermore, at the same time that data retrieval from ENS160 stops, the temperature starts to drop again which implies that the ENS160's heaters are turned off.

Looking back at the log output from the ENS160 which is produced when the sensor does not provide data anymore, I've noticed that it reports
````
[17:55:36][V][ens160:157]: Status: ENS160 STATAS bit    0x0
````
which - according to the [datasheet](https://dfimg.dfrobot.com/nobody/wiki/66d326ec8ba3da553df93280d09e6b15.pdf) would imply that there is no operating mode selected:
[image]

I've got the feeling that something causes the ENS160 to forget it's operating mode and the current esphome implementation of the sensor does not re-configure the sensor after the initial setup phase.

Maybe @vincentscode who implemented the ENS160 support can have a look at this? I am also unsure as to whether I should open a new issue for these problems.

Best wishes & thanks for your help!

hjeldin, 2023-12-05:
I'm having the same problem:
```
[16:04:49][V][ens160:157]: Status: ENS160 STATAS bit    0x0
[16:04:49][V][ens160:159]: Status: ENS160 STATER bit    0x0
[16:04:49][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[16:04:49][V][ens160:162]: Status: ENS160 NEWDAT bit    0x0
[16:04:49][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
[16:04:49][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[16:04:53][V][bme280.sensor:206]: Sending conversion request...
[16:04:53][V][bme280.sensor:238]: Got temperature=21.8°C pressure=1009.1hPa humidity=38.8%
[16:04:53][V][sensor:043]: 'BME280 Temperature': Received new state 21.770000
[16:04:53][D][sensor:094]: 'BME280 Temperature': Sending state 21.77000 °C with 1 decimals of accuracy
[16:04:53][V][sensor:043]: 'BME280 Pressure': Received new state 1009.123535
[16:04:53][D][sensor:094]: 'BME280 Pressure': Sending state 1009.12354 hPa with 1 decimals of accuracy
[16:04:53][V][sensor:043]: 'BME280 Humidity': Received new state 38.839844
[16:04:53][D][sensor:094]: 'BME280 Humidity': Sending state 38.83984 % with 1 decimals of accuracy
[16:04:53][W][component:214]: Component bme280.sensor took a long time for an operation (0.06 s).
[16:04:53][W][component:215]: Components should block for at most 20-30ms.
```

This doesn't seem to be a component issue but rather a [firmware/hardware issue](https://github.com/sciosense/ENS160_driver/issues/8) with the ENS160

I tried fetching the data from the general purpose registers assuming that the chip was "stuck" waiting for someone to read the data, but i only managed to reset the NEWGPR flag.

edwardtich1, 2023-12-06:
`
20:17:24 | [I] | [i2c.arduino:075] | Found i2c device at address 0x38
-- | -- | -- | --
20:17:24 | [I] | [i2c.arduino:075] | Found i2c device at address 0x53
20:17:24 | [C] | [aht10:137] | AHT10:
20:17:24 | [C] | [aht10:138] | Address: 0x38
20:17:24 | [E] | [aht10:140] | Communication with AHT10 failed!

edwardtich1, 2023-12-07:
detect two devices
ens160 and unknown
but not AHT10
then what is the second device?

ironsilk, 2023-12-16:
I'm getting the same with esp32 + ENS160&AHT21 chip:
```
[14:14:10][V][ens160:157]: Status: ENS160 STATAS bit    0x0
[14:14:10][V][ens160:159]: Status: ENS160 STATER bit    0x0
[14:14:10][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[14:14:10][V][ens160:162]: Status: ENS160 NEWDAT bit    0x0
[14:14:10][V][ens160:164]: Status: ENS160 NEWGPR bit    0x1
[14:14:10][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
```

I've probably tried in a lot of ways, the init looks promising but data never comes out

```
[11:57:32][D][ens160:300]: Setup successful
[11:57:32][I][ens160:304]: Firmware Version: 5.4.6
[11:57:32][C][ens160:306]:   Address: 0x53
[11:57:32][C][ens160:307]:   Update Interval: 60.0s
[11:57:32][C][ens160:308]:   CO2 Sensor: 'eCO2 Value'
[11:57:32][C][ens160:308]:     Device Class: 'carbon_dioxide'
[11:57:32][C][ens160:308]:     State Class: 'measurement'
[11:57:32][C][ens160:308]:     Unit of Measurement: 'ppm'
[11:57:32][C][ens160:308]:     Accuracy Decimals: 0
[11:57:32][C][ens160:308]:     Icon: 'mdi:molecule-co2'
[11:57:32][C][ens160:309]:   TVOC Sensor: 'Total Volatile Organic Compound'
[11:57:32][C][ens160:309]:     Device Class: 'volatile_organic_compounds'
[11:57:32][C][ens160:309]:     State Class: 'measurement'
[11:57:32][C][ens160:309]:     Unit of Measurement: 'ppb'
[11:57:32][C][ens160:309]:     Accuracy Decimals: 0
[11:57:32][C][ens160:309]:     Icon: 'mdi:radiator'
[11:57:32][C][ens160:310]:   AQI Sensor: 'Air Quality Index'
[11:57:32][C][ens160:310]:     Device Class: 'aqi'
[11:57:32][C][ens160:310]:     State Class: 'measurement'
[11:57:32][C][ens160:310]:     Unit of Measurement: 'index'
[11:57:32][C][ens160:310]:     Accuracy Decimals: 0
[11:57:32][C][ens160:310]:     Icon: 'mdi:chemical-weapon'
[11:57:32][C][ens160:313]:     Temperature Compensation: 'AHT21 Temperature'
[11:57:32][C][ens160:313]:     Device Class: 'temperature'
[11:57:32][C][ens160:313]:     State Class: 'measurement'
[11:57:32][C][ens160:313]:     Unit of Measurement: '°C'
[11:57:32][C][ens160:313]:     Accuracy Decimals: 2
[11:57:32][C][ens160:314]:     Humidity Compensation: 'AHT21 Humidity'
[11:57:32][C][ens160:314]:     Device Class: 'humidity'
[11:57:32][C][ens160:314]:     State Class: 'measurement'
[11:57:32][C][ens160:314]:     Unit of Measurement: '%'
[11:57:32][C][ens160:314]:     Accuracy Decimals: 2
```
Thought it might be something wrong with the wiring but the AHT is sending data just fine, tried in both 3.3v and 5v.

galvheim, 2023-12-17:
From what I have tested and found out, it needs considerable heat to work.  Tried many times balsting it with some hot air or infra heat and it will quickly start work. Inslutaing it may be a solution and then calibrate the AHT10 later.

edwardtich1, 2023-12-20:
09:13:17 | [W] | [ens160:244] | Invalid external temperature - compensation values not updated
-- | -- | -- | --

hjeldin, 2023-12-20:
I can confirm that in my specific case the issue was a missing decoupling capacitor on the VDD rail. Adding a 100nF fixed the chip brownout issue and the library now works correctly.

edwardtich1, 2023-12-20:
> Я могу подтвердить, что в моем конкретном случае проблема заключалась в отсутствии развязывающего конденсатора на шине VDD. Добавление 100 нФ устранило проблему с понижением напряжения чипа, и теперь библиотека работает корректно.

Could you be more specific about what needs to be done in the diagram?

thank you in advance!

galvheim, 2023-12-20:
> I can confirm that in my specific case the issue was a missing decoupling capacitor on the VDD rail. Adding a 100nF fixed the chip brownout issue and the library now works correctly.

Missing or added? More info on this would be interesting.

hjeldin, 2023-12-20:
> Could you be more specific about what needs to be done in the diagram?
> 
> thank you in advance!

> Missing or added? More info on this would be interesting.

Check out [here](https://github.com/sciosense/ENS160_driver/issues/8)

edwardtich1, 2023-12-21:
> > Можно поподробнее, что нужно сделать на схеме?
> > заранее спасибо!
> 
> > Отсутствует или добавлен? Дополнительная информация об этом была бы интересна.
> 
> Проверьте [здесь](https://github.com/sciosense/ENS160_driver/issues/8)

I added the capacitor from this link

nothing changed

12:21:24 | [W] | [ens160:244] | Invalid external temperature - compensation values not updated
-- | -- | -- | --

terryuland, 2024-01-02:
> I'm getting the same with esp32 + ENS160&AHT21 chip:
>
> ```
> [14:14:10][V][ens160:157]: Status: ENS160 STATAS bit    0x0
> [14:14:10][V][ens160:159]: Status: ENS160 STATER bit    0x0 [...]

This is exactly what I'm getting, using ESP8266 Wemos D1 Mini. It's the only sensor connected and yes I did wait a few hours and never got any non-zero data. Have only tried 3.3v though (wiring 3.3v out from ESP8266 to the sensor, MCU is powered via USB)

galvheim, 2024-01-02:
Heat it up... You will see numbers.

terryuland, 2024-01-02:
> Heat it up... You will see numbers.
That didn't work, however, I did make it work by connecting a 100uF cap from ground to the 1.8v rail as mentioned here: https://github.com/sciosense/ENS160_driver/issues/8

Looks like a manufacturing defect.

terryuland, 2024-01-03:
I would like sleep mode integrated into ESPHome. The sensor itself does support deep sleep, where current @ 3.3v goes from around 30mA in normal operation to 0.03mA (30 micro amps) in deep sleep mode. In order to activate this function the OPMODE must be set to 0x0 in which case the sensor will immediately go to sleep. Setting OPMODE back to standard (0x02) will wake it up, but there must be a delay of at least one second until data is valid. 

The Adafruit Arduino/ESP32 library almost implements this, sleep is defined in the header file, but never actually used in the C source, though in a sketch you can easily do ens160.setMode(ENS160_OPMODE_DEP_SLEEP) which will put it to sleep and the same setMode method using ENS160_OPMODE_STD to wake it up. 

However I do not see a way to do that with the existing ESPHOME integration as I have no idea how to directly run code from YAML files (I admit I am very new to ESPHOME). So, maybe someone who knows how to do that can provide a workaround for turning on sleep or integrate it into the existing library.

kbenton, 2024-02-08:
As others have mentioned, you will almost certainly need to put a capacitor between the 1.8v rail and ground. Note that some boards are subtly different. I've attached a picture showing two different boards I got that required different solder points. I posted on this on the ESPHome discord as well.

Having gotten them to reliably send data, I'm now having the problem that I don't know if I can trust that data. I built three identical circuits, presently all on breadboards, but otherwise the components are the same (except for the two slightly different manufacturers of the ens160+aht21 boards). They're all sitting next to each other on my bench and they provide *wildly* different numbers. One will report TVOC of 200, the second 400, the third 1800. Then it'll be 400, 92, 1200. I suspect the third board is genuinely defective, but even so there's a ton of variation that I can't easily explain. It's one thing that the temperature data are high by between 4 and 10 degrees F... it's at least consistently so and I could make corrections. But the VOC and eCO2 numbers are just all over the place.

These setups have SGP41 and PM2.5 sensors as well, and those are all in very close agreement, as I would expect, so I don't think there's a big flaw in my general arrangement. Sadly, I don't think these boards are a great design and I may have to shop for alternatives, which sucks because they're not particularly cheap. I kinda think they should've just left the ENS160 and AHT21 as separate small boards. These do exist, but solo ENS160 ones are a lot less available than these combo boards and tend to cost more, but that may be because they actually work :/

[image]

terryuland, 2024-02-08:
> As others have mentioned, anyone using this type of board will almost certainly need to put a capacitor between the 1.8v rail and ground. Note that some boards are subtly different. I've attached a picture showing two different boards I got that had slight differences and required different solder points. I posted on this on the ESPHome discord as well.
> 
> Having gotten them to reliably send data, I'm now having the problem that I don't know if I can trust that data. I built three identical circuits, presently all on breadboards, but otherwise the components are the same (except for the two slightly different manufacturers of the ens160+aht21 boards). They're all sitting next to each other on my bench and they provide _wildly_ different numbers. One will report TVOC of 200, the second 400, the third 1800. Then it'll be 400, 92, 1200. I suspect the third board is genuinely defective, but even so there's a ton of variation that I can't easily explain. It's one thing that the temperature data are high by between 4 and 10 degrees F... it's at least consistently so and I could make corrections. But the VOC and eCO2 numbers are just all over the place.
> 
> These setups have SGP41 and PM2.5 sensors as well, and those are all in very close agreement, as I would expect, so I don't think there's a big flaw in my general arrangement. Sadly, I don't think these boards are a great design and I may have to shop for alternatives, which sucks because they're not particularly cheap. I kinda think they should've just left the ENS160 and AHT21 as separate small boards. These do exist, but solo ENS160 ones are a lot less available than these combo boards and tend to cost more, but that may be because they actually work :/
> 
> ![ENS160boards](https://private-user-images.githubusercontent.com/139949/303224957-86231e1a-b12a-48e5-8014-1b70eed984d6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0Mjc2NzMsIm5iZiI6MTcwNzQyNzM3MywicGF0aCI6Ii8xMzk5NDkvMzAzMjI0OTU3LTg2MjMxZTFhLWIxMmEtNDhlNS04MDE0LTFiNzBlZWQ5ODRkNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwOFQyMTIyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZDMxOGQ3ZmFlMTFmZGU5OWQyNDUzMTE4YzUyNDFiZjYzNzRmODI5MWZmZGNiNWI2NDBlZmE2ZGIyZmIzMWU1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BFto9CBDS8iQz8BloUu9N3vndc_YaBI163tGSQSa2iA)

Yea I've noticed the accuracy issues too, particularly, it looks like the AHT21 temp is always around 7-8 degrees F, at least compared to a DHT sensor on the same prototyping board. Someone mentioned that could be from the heaters in the ENS160 but I would think you'd easily notice if the chip was hot enough to heat the air all the way over to the AHT21 by that much. Regarding the VOCs, I have a couple of Alexa AQMs and have seen wild stuff with them as well. I think VOCs tend to form these ribbons in the air where one sensor will literally be right beside another, but have greatly different readings based on the air currents.  

Generally speaking though looking over my history in HA I can see changes over time and identify when I was in the room, when a window or doors were open, and so forth. I do find it odd that eCO2 and VOC seem to track so closely.

kbenton, 2024-02-08:
Interesting. I don't mind so much if the numbers are jumpy. In my case one of the three is clearly well off from the others. I also wonder how much it matters that they were all calibrated at different times... i suspect that could be a big part of it. If one was calibrated in pretty clean air and the other two in much worse air, that first might *always* report significantly higher in the same conditions.

I did some searching for whether it's possible to reset the calibration, so I could baseline all three at the same time, but I didn't find much.

I have some independent AHT21 board, so I've gone ahead and ordered 3 standalone ENS160 boards from a different vendor. Once they get here from china, I'll take a stab at re-wiring to use those, and fire them all up at the same time. Since the calibration data is on-chip for ENS160s, they should all start calibrating in the same environment, and since the I2C addresses are the same, i shouldn't have to make any changes to the ESPHome configuration at all.

We'll see. I'm even considering putting them in a large, sealed plastic bin with some activated charcoal for a few hours before turning them on for the first time to make the environment both clean as clean and homogenous as possible during the initial calibration period.