esphome/feature-requests #2036: rp2040: http_request

janneman001, 2022-12-24:
**Describe the problem you have/What new integration you would like**

With esp devices i can use http_requests. On rp2040 not (yet)

**Please describe your use case for this integration and alternatives you've tried:**

I would like to use something like this. I call it on_value

script:
  id: metrics_2_influx
  then:
    # send data to influxdb
    - http_request.post:
        url: https://metrics.exampledomain.com/influxdb/write?db=metrics
        headers: 
          Content-Type: text/plain; charset=utf-8
          Authorization: !secret metrics_auth
        body: !lambda |-
          char temperature[64];
          sprintf(temperature, "%0.1f", id(room_temp).state);
          return ((std::string) "esphome,computer=picow,sensor=room “+
          “room_temp="+temperature).c_str();
          }

The workaround i try now is to use an automation via the integration in home assistant.
However this is not efficient for data processing.

**Additional context**

esphome is a great tool. I like the new addition of rp2040 to it. 
I would like to have this added if not already planned.

siaub, 2023-05-10:
+1 for this request, it's a shame there is still no support yet

kpfleming, 2024-06-26:
This is supported in 2024.6, so this request can be closed.

closed 2024-06-26 by nagyrobi