esphome/feature-requests #2040: support for "Adafruit LC709203F LiPoly / LiIon Fuel Gauge and Battery Monitor - STEMMA JST PH & QT / Qwiic"

cTurtle98, 2022-12-26:
**Describe the problem you have/What new integration you would like**

I would like support for reading battery status with this sensor [Adafruit LC709203F LiPoly / LiIon Fuel Gauge and Battery Monitor - STEMMA JST PH & QT / Qwiic](https://www.adafruit.com/product/4712)

**Please describe your use case for this integration and alternatives you've tried:**

The adafruit feather board that I have has this chip built in for monitoring the battery on the feather battery port. (see additional context)

I have not tried any alternatives. I would like to be able to use built in features.

**Additional context**

I have an [Adafruit ESP32-S2 Feather with BME280 Sensor - STEMMA QT - 4MB Flash + 2 MB PSRAM](https://www.adafruit.com/product/5303) and it has one of those sensors onboard for the lipo port

Source:
```
[13:54:33][I][i2c.arduino:068]: Results from i2c bus scan:
[13:54:33][I][i2c.arduino:074]: Found i2c device at address 0x0B
[13:54:33][I][i2c.arduino:074]: Found i2c device at address 0x77
```
https://learn.adafruit.com/i2c-addresses/the-list says that that `0x0B` is the `LC709203` (0x77 is the bme280, an already supported chip)

as well as

https://www.adafruit.com/product/5303 says
> Features:
> LiPoly battery monitor - LC709203 chip actively monitors your battery for voltage and state of charge / percentage reporting over I2C

cTurtle98, 2023-01-01:
here's the adafruit library for using the device in arduino https://github.com/adafruit/Adafruit_LC709203F

cTurtle98, 2023-01-01:
adafruit article about using the sensor in arduino https://learn.adafruit.com/adafruit-esp32-s2-feather/lc709203-simple-data

joshspicer, 2023-09-02:
Ever resolve this?  I'd like to do the same thing

def1149, 2023-10-05:
Has support for the LC709203F been incorporated? clicking through various lines on this page it looks like someone developed code for it.

ilikecake, 2025-03-23:
Not sure if anyone is still interested in this, but I have submitted a [PR](https://github.com/esphome/esphome/pull/8037) to have this component added.

closed 2025-06-01 by kbx81