esphome/feature-requests #2043: Ability to archive devices (offline ones for example)

HollyFredD, 2022-12-29:
Many of my devices are offline for various reasons and it would be nice to be able to archive devices to hide them for the main dashboard. (useful for the update process to avoid getting errors)

It would also be important to be able to un-archive them once they are installed/used again.

pnichols77, 2023-01-04:
Just googled to see if this was possible and found this recent feature request so adding my voice in support. I'd like to be able to archive devices that are no longer active - either because they're not being used, or have been repurposed - but without losing easy access to the yaml that was used with them. I could of course move the yaml files out and delete but that introduces a layer of management that could be easily solved with an archive option and a filter on the esphome dashboard.

erik3rik, 2023-04-04:
I also would like this feature. Some of my devices are used only in some seasons (e.g. like an automated advent candle) and that one is disconnected 11 months per year. There is no point in showing it, nor (try to) update it (and fail) when updating all devices etc. Some old stuff is there in case I need the code etc, but will never be online again. I would like to keep them, but have the ability to hide/ignore them.

solideus, 2023-11-06:
I like this feature too

jernauH, 2023-11-27:
I have the same issue with certain devices only used for holidays or just keeping the code for future use so +1 for an archive feature or ability to hide devices not being used

andreas-bulling, 2025-05-03:
This feature request was added back in December 2022, i.e. 2.5 years ago.

How can this not yet be an ESPHome feature?

The dashboard hasn't seen many updates in the last 1-2 years. And the one it has seen were mainly cosmetic.

The dashboard is what users interact with on a daily basis. What potential/new users will see first when they decide whether they want to give ESPHome a try.

I don't get why it's apparently at such low priority.

wenestvedt, 2026-08-20:
I, too, would like to be able to retain unused config files. 

If I am going to work fully within ESPHome's integration for Home Assistant, I need to be able to manage config files there (without, e.g., exporting them to text files on some other host -- which would lose the current build files).

If there's already a recommended way to handle this -- like moving the files with the File Editor -- emphasizing that fact in the docs would be great.

Thanks!