esphome/feature-requests #2046: Add support for Fluvius V1.7.1 DSMR messages (peak tarrif Belgium)

TheGroundZero, 2023-01-01:
**Describe the problem you have/What new integration you would like**

Add sensors related to the peak tarrif in Belgium.

See https://github.com/ndokter/dsmr_parser/commit/d05fe2692bce9da5bca4103008a6327b4901106a

**Please describe your use case for this integration and alternatives you've tried:**

Peak tarrif is based on the highest peak per day, where peaks are calculated based on the average power consumption over 15 minutes.
The utility provider that is responsible for installing the digital energy meters, Fluvius, has added extra fields to the DSMR datastream that hold data on the current 15min avg, highest peak for the month and highest peak over 13 months.

**Additional context**

https://m.facebook.com/groups/makeinbelgium/permalink/5788013861284159/

mhostyn, 2023-01-09:
For those intrested, i've been messing around in the code and found these points of intrest
Changes i made to glmnet/arduino-dsmr
https://github.com/glmnet/arduino-dsmr/compare/master...mhostyn:arduino-dsmr:master
And the changes made to esphome
https://github.com/esphome/esphome/compare/dev...mhostyn:esphome:dev

it does seem to detect the actual value
[image]
however it does also break everything else, so i hope this may be enough of a push for anyone else who is waiting

cosmos1978, 2023-01-30:
Me and all other belgian users are waiting for this

cosmos1978, 2023-02-02:
Someone already implemented it. link is in this topic.
https://github.com/esphome/issues/issues/3995

I compiled it and tested it seems to work as expected.

mhostyn, 2023-02-02:
> Someone already implemented it. link is in this topic. [esphome/issues#3995](https://github.com/esphome/issues/issues/3995)
> 
> I compiled it and tested it seems to work as expected.

Yeah, sorry it does indeed work however at the time i had exams, so selfish excuses aside... i am using my own fork in which i manually imported the source code of [glmnet/arduino-dsmr](https://github.com/glmnet/arduino-dsmr/compare/master...mhostyn:arduino-dsmr:master) directly into my own fork of [mhostyn/esphome]https://github.com/mhostyn/esphome)

The original error i had was because i did not know anything about the original code yet and i did not set the 1.6.0 to parse as a timestamp-string/value

for now pushing or merging the nicely formatted code by @klenaers in https://github.com/esphome/issues/issues/3995#issuecomment-1374179663_ into the actual repositories
(For current q-hourly usage (the 'Sawtooth') and current monthly max)

I haven't looked at the code much further but the 13month average seems to have a (start-timestamp/end-timestamp/value) which might not yet be implemented in the parser used by [glmnet/arduino-dsmr/](https://github.com/glmnet/arduino-dsmr/) https://github.com/glmnet/arduino-dsmr/

Just as a reference (and another proof of concept) these are the configurations that work for me at the moment (or rather the additions to an already working configuration) this does not include the 13month average

`
external_components:
  - source:
      type: git
      url: https://github.com/mhostyn/esphome
      ref: dsmr-testing
    refresh: 0s
    components: [ dsmr ]

sensor:
  - platform: dsmr
    quart_hourly_current_average_peak_consumption:
      name: ${friendly_name} Current Quart-hourly consumption
    quart_hourly_max_peak_this_month:
      name: ${friendly_name} Quart-hour Monthly max
    # quart_hourly_average_peak_last_13months:
    #   name: ${friendly_name} Quart-hour Yearly avg

`

as a personal note, adding these "Peak at Specific time" values might also be able to use a extra text_sensor:

cosmos1978, 2023-02-02:
In order to get the 13 month peak to work I included this sensor in homeassistant.

      peak_current_year:
        friendly_name: Peak current year
        value_template: >-
          {{ states('sensor.peak_power_consumption_last_13_months_belgium').split(')(')[-1] | replace('*kW)',' kW') }}

Use the sensor name you used in your esphome.yaml file and you should be good.

It only shows the value of the current year. The meter will have 2 values however one for previous year and one for the current year.  

example.
(2)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221207221500W)(12.948*kW)(230201000000W)(230126000000W)(11.745*kW)

So this sensor will show the last value 11.754 kW

cosmos1978, 2023-02-08:
> > Someone already implemented it. link is in this topic. [esphome/issues#3995](https://github.com/esphome/issues/issues/3995)
> > I compiled it and tested it seems to work as expected.
> 
> Yeah, sorry it does indeed work however at the time i had exams, so selfish excuses aside... i am using my own fork in which i manually imported the source code of [glmnet/arduino-dsmr](https://github.com/glmnet/arduino-dsmr/compare/master...mhostyn:arduino-dsmr:master) directly into my own fork of [mhostyn/esphome]https://github.com/mhostyn/esphome)
> 
> The original error i had was because i did not know anything about the original code yet and i did not set the 1.6.0 to parse as a timestamp-string/value
> 
> for now pushing or merging the nicely formatted code by @klenaers in [esphome/issues#3995 (comment)](https://github.com/esphome/issues/issues/3995#issuecomment-1374179663)_ into the actual repositories (For current q-hourly usage (the 'Sawtooth') and current monthly max)
> 
> I haven't looked at the code much further but the 13month average seems to have a (start-timestamp/end-timestamp/value) which might not yet be implemented in the parser used by [glmnet/arduino-dsmr/](https://github.com/glmnet/arduino-dsmr/) https://github.com/glmnet/arduino-dsmr/
> 
> Just as a reference (and another proof of concept) these are the configurations that work for me at the moment (or rather the additions to an already working configuration) this does not include the 13month average
> 
> ```yaml
> external_components:
>   - source:
>       type: git
>       url: https://github.com/mhostyn/esphome
>       ref: dsmr-testing
>     refresh: 0s
>     components: [ dsmr ]
> 
> sensor:
>   - platform: dsmr
>     quart_hourly_current_average_peak_consumption:
>       name: ${friendly_name} Current Quart-hourly consumption
>     quart_hourly_max_peak_this_month:
>       name: ${friendly_name} Quart-hour Monthly max
>     # quart_hourly_average_peak_last_13months:
>     #   name: ${friendly_name} Quart-hour Yearly avg
> ```
> 
> as a personal note, adding these "Peak at Specific time" values might also be able to use a extra text_sensor:

Just noticed that your code got pushed to the official github glbnet.

But you did not include the 13 month peak. Could you push an update to include it as a string since it is really easy to fetch the data from it straight from homeassistant. Then the only thing remaining is for someone to push the esphome code from [https://github.com/klenaers](url) to esphome

bartv, 2023-02-14:
There is a PR for this now https://github.com/esphome/esphome/pull/4412

bwynants, 2023-02-16:
and parsing for third field https://github.com/glmnet/arduino-dsmr/pull/15

bwynants, 2023-02-18:
`external_components:
  - source:
      type: git
      url: https://github.com/bwynants/esphome
      ref: capacity_test
    components: [dsmr]
    refresh: 0s
`

`sensor:
  - platform: dsmr
    active_energy_import_current_average_demand:
      name: "Current Average Quarterly Demand"
    active_energy_import_maximum_demand_running_month:
      name: "Month Maximum Quarterly Demand"
    active_energy_import_maximum_demand_last_13_months:
      name: "Last 13 Maximum Quarterly Demand"
`

bwynants, 2023-02-18:
```
external_components:
  - source:
      type: git
      url: https://github.com/bwynants/esphome
      ref: capacity_test
    components: [dsmr]
    refresh: 0s

```

```
sensor:
  - platform: dsmr
    active_energy_import_current_average_demand:
      name: "Current Average Quarterly Demand"
    active_energy_import_maximum_demand_running_month:
      name: "Month Maximum Quarterly Demand"
    active_energy_import_maximum_demand_last_13_months:
      name: "13 Month Maximum Quarterly Demand"

```

TheGroundZero, 2023-03-29:
@KLucky13 Unless you opt for monthly invoicing or dynamic tarrif, the invoice you get after a year makes use of a consumption profile.
I.e., they don't sum your actual consumption per day/week/month but take your total yearly consumption and apply a model that reflects the average consumption of all households and which gives a percentage of the total for every specific month. Sort of like a bell-curve with less consumption in the summer and more in the winter.
https://www.vreg.be/nl/jaar-en-maandafrekening

KLucky13, 2023-03-29:
I have a monthly invoice at the moment and pay what I actually consumed

TheGroundZero, 2023-05-03:
Does anyone know when this will be implemented in the main repo?

The specs can be found here: https://maakjemeterslim.be/rails/active_storage/disk/eyJfcmFpbHMiOnsibWVzc2FnZSI6IkJBaDdDRG9JYTJWNVNTSWhlblJ4YW14cU5XdHRhalZsWkRWbmJIaDNNak4zYmpReGFUazRkd1k2QmtWVU9oQmthWE53YjNOcGRHbHZia2tpVzJGMGRHRmphRzFsYm5RN0lHWnBiR1Z1WVcxbFBTSmxMVTFWUTFOZlVERmZSV1JmTVY4M1h6RXVjR1JtSWpzZ1ptbHNaVzVoYldVcVBWVlVSaTA0SnlkbExVMVZRMU5mVURGZlJXUmZNVjgzWHpFdWNHUm1CanNHVkRvUlkyOXVkR1Z1ZEY5MGVYQmxTU0lVWVhCd2JHbGpZWFJwYjI0dmNHUm1CanNHVkE9PSIsImV4cCI6IjIwMjMtMDUtMDNUMDc6MjM6NDAuMzEyWiIsInB1ciI6ImJsb2Jfa2V5In19--9d95fc582964468fd9955b0ce9c50767a4025ede/e-MUCS_P1_Ed_1_7_1.pdf

Marked in yellow are the changes added by v1.7.1 of the spec, which are the 3 fields for the max. demand
[image]

bwynants, 2023-07-17:
> Does anyone know when this ([esphome/esphome#4412](https://github.com/esphome/esphome/pull/4412)) will be implemented in the main repo? @glmnet ?

https://github.com/esphome/esphome/pull/5081 is merged and will be in ESPHome 2023.7 i think

lieven121, 2023-08-18:
Afaik it has, i have peak tarif now and commented the link to the external code.

It compiles and works.

So then the question is why is this issue still open ?

TheGroundZero, 2023-08-18:
Looks like esphome#5011 added the functionality, but the documentation is still lacking the added sensors?

Am I right this needs to be added?
```yaml
sensor:
  - platform: dsmr
    # ...
    active_energy_import_current_average_demand:
      name: "Current Average Quarterly Demand"
    active_energy_import_maximum_demand_running_month:
      name: "Month Maximum Quarterly Demand"
    active_energy_import_maximum_demand_last_13_months:
      name: "13 Month Maximum Quarterly Demand"
```

closed 2023-08-20 by jesserockz