esphome/feature-requests #2047: Support of water heaters

Bascht74, 2023-01-03:
**Describe the problem you have/What new integration you would like**
Home Assistant has an integration for water heaters: 
https://www.home-assistant.io/integrations/water_heater/

I would like to use them within ESPHome as well. 
They are similar to climate devices but don't control a room temperature, but a water temperature

**Please describe your use case for this integration and alternatives you've tried:**
Control my water heater that is connected to Mobus via modbus controller as a template water heater

**Additional context**
./.

randybb, 2023-01-03:
I think it is used only for the heaters that are directly communicating with HA.
I am controlling my heater/water heater via esphome and I am using the standard climate component (one component for heater and one for water heater).

MiguelAngelLV, 2023-08-30:
Home Assistant will add new tiles card for Water Heater. I think could be very interesting have a especific water heater type in EspHome

barndawgie, 2023-09-16:
For the esphome-econet integration, we currently use a Climate entity to represent a Water Heater. Being able to use the HA Water Heater entity would be interesting for sure!

kpfleming, 2024-07-04:
I've begun working on this, PRs will be linked to this feature request.

Water heater support in HA is extremely basic at the moment, it's not possible to report the actual heating mode of a hybrid unit, for example.

barndawgie, 2024-07-04:
> Water heater support in HA is extremely basic at the moment, it's not possible to report the actual heating mode of a hybrid unit, for example.

Yes, I've been looking at it and am honestly not sure we'd move ESPHome-EcoNet as I think we'd lose a bunch of functionality with the current HA implementation.

kpfleming, 2024-07-04:
I can't tell for sure, but I don't think there would be much functionality loss. The water_heater platform in HA supports many of the necessary modes (electric, heat_pump, etc.) which are currently 'custom presets' in ESPHome-EcoNet. It's certainly worth having a conversation on the issue over there to document which bits of functionality are needed.

dhoeben, 2025-12-03:
> > Hi all, I just created a pull request regarding this. Hopefully it will pass and feel free to check it out and let me know where to improve.
> 
> @dhoeben are you planning to prepare a new pull request with water heater component? I see that the current PR status is Closed.

Yep, but very busy at the moment. I think I will publish beginning next year, I will keep you updated!