esphome/feature-requests #2048: Tunai Bluetooth Button

h0jeZvgoxFepBQ2C, 2023-01-04:
**Describe the problem you have/What new integration you would like**
I would like to add my Tunai Bluetooth Button via esphome bluetooth proxy to my HA installation

**Please describe your use case for this integration and alternatives you've tried:**

I've tried to add it, but somehow it's not recognized in HA.

**Additional context**

[image]

But during pairing it shows following error/warning:
[image]

Please tell me if/how I can help to make it work! ❤️ Thank you so much for esphome, this is really a wonderful project which I use everyday! ❤️

nagyrobi, 2023-01-04:
Pairing is not yet supported.

h0jeZvgoxFepBQ2C, 2023-01-04:
Ah I see, not even if there is no PIN required? I just press the +/- buttons for 5 seconds and it switches to pairing mode.
Just to understand it: active bluetooth is supported, but not with pairing I guess?

Thanks for your fast answer!

nagyrobi, 2023-01-04:
Not aware of being supported. Please ask in our support channel on Discord for further details.