esphome/feature-requests #2049: WS2814 / WS2814A addressable led chip support for NeopixelBus Light

Kai-X1, 2023-01-05:
**Describe the problem you have/What new integration you would like**

Add support for WS2814 / WS2814A RGBW addressable LED chips in NeopixelBus Light

**Please describe your use case for this integration and alternatives you've tried:**

Id like to be able to use WS2814A addressable LED chips with the NeopixelBus Light, I've tried various settings for other chips to see if any works. but from what I gather WS2814 is 32bit, where as others are only 24bit. 

thank you

**Additional context**

Datasheets only seem to be in Chinese so far but maybe someone knowledgeable might be able to interpret the data from the tables.
https://datasheet.lcsc.com/lcsc/2012110135_Worldsemi-WS2814_C965562.pdf
https://datasheet.lcsc.com/lcsc/2111171930_Worldsemi-WS2814A_C2920044.pdf

thomasklingbeil, 2023-03-07:
I tried it out and it actually works with the following settings:

```
light:
  - platform: neopixelbus
    type: WRGB
    variant: WS2811
```

manundi, 2023-08-05:
> I tried it out and it actually works with the following settings:
> 
> ```
> light:
>   - platform: neopixelbus
>     type: WRGB
>     variant: WS2811
> ```

For me this setup shows wrong colors: red = green etc.

thomasklingbeil, 2023-08-05:
> For me this setup shows wrong colors: red = green etc.

And if you use "type: WGRB" instead?

Kinggrass, 2024-07-04:
@Kai-X1
Got it working?
For me not even a light turn on.

evelant, 2024-07-14:
Does anybody have a working config for ws2814? Sorry I'm new to all of this. Got some F[COB addressable 24v strips that use ws2814 ](https://www.aliexpress.us/item/3256805811194861.html?spm=a2g0o.order_detail.order_detail_item.4.6f6439d3Qsvbta&gatewayAdapt=glo2usa) and am having trouble controlling them. With the config suggested by @thomasklingbeil above only half the strip lights up and it's in seemingly random colors.

CreaValix, 2024-07-26:
Good morning @evelant, yes, I do have the same configuration as @thomasklingbeil. Keep in mind that ws2814 is 24 V and there are six LEDs in series.

I have a 5 m LED strip vom BTF-Lighting, BTF-24V-060L-W, which has 60 LEDs per meter. So I set num_leds to 60 × 5 ÷ 6 = 50. I tried variants WS2811 and WS2813. It seems to work with both.

This is my working configuration for an esp8266 ESP8266 ESP-01 1M board:

light:
  - platform: neopixelbus
    name: Schlafzimmer Deckenbeleuchtung
    type: WRGB
    variant: WS2811
    pin: GPIO2
    num_leds: 50
    method:
      type: esp8266_uart
      bus: 1
      async: false
    default_transition_length: 0s

I still do get flickering during the transitions. It was really bad with 3,2 V ESP input voltage. There is almost no flickering with 3,4 V. Above linked datasheet (thanks!) says that D_IN should be 5 V. So I'll need a logic shifter.

evelant, 2024-07-26:
Thanks, I got it working with a similar config. Turns out I had wired it wrong. After fixing the wiring (needed a common ground for the strip and the esp32!) it all works beautifully.