esphome/feature-requests #2057: It would be very useful if the "<<: !include file.yml" would have a more fined grained inheritance, instead of the "last one wins" behaviour

lucius-the, 2023-01-13:
**Describe the problem you have/What new integration you would like**
I have an `<<: !include base.yaml` directive in my esphome config files. That include contains settings to report uptime and Wifi RSSI for all my nodes, suppose the `base.yml` looks like this:
```
sensor:
  - platform: uptime
    name: "${node_name}_uptime"
    update_interval: 1h
  - platform: wifi_signal
    name: "${node_name}_rssi"
    update_interval: 60s
```
Then, all my node config files contain the `<<: !include base.yaml` line far up on top, right after substituitions, to include all from base.yml.
However, if my node config files also contain a `sensor:` entry, then anything from `sensor:` in base.yml doesn't get included in the build at all. Actually, depends on where I put include line in the node config file - basically I can choose to either loose sensors from base config (putting config higher on top) or loose sensors from the main node config (putting include lower to the bottom in main config file).

I find this kind of unexecpected and quite limiting. I can't really share (merge) common stuff, when, as soon as same root node config item appears, it completely nulls all the perviously included ones. Well, that's not "inheritance". An include should, in my opinion, always include and then if something exists again it should override - but it should do so much more granuraly. For example, I soulh be able to override "update_interval" in my node config, not destroy everything from "sensor:" just because another "sensor:" entry appeard. So, a behaviour more in line with class inheritance behaviour in OOM programming would be great (true inheritance, with overriding) would be much, much more useful than current state. Current state is more like "I created a new object", which means "I'm not inheriting anything" because I'm emtpy.

I hope my explanation makes sense...

**Please describe your use case for this integration and alternatives you've tried:**
Sharing common config entries more effectively.

**Additional context**

Thank you very much for all great work !

bkaufx, 2023-01-13:
https://esphome.io/guides/configuration-types.html#local-packages

lucius-the, 2023-01-14:
Thank you very much Brian ! will try soon

lucius-the, 2023-01-14:
Yes ! That works.
Simply replacing this:
```
<<: !include base.yaml
```
with this:
```
packages:
  device_base: !include base.yaml
```

Made it behave in line with my expectations :)
I didn't even have to define separate sensor: and switch: includes as in the example in above link - I just put everything in one file and it works fine.

Thanks a lot ! Closing the issue.

closed 2023-01-14 by lucius-the