esphome/feature-requests #2058: Ability to protect the button entity with a PIN code/password

lucius-the, 2023-01-13:
**Describe the problem you have/What new integration you would like**
I know I can get around this with tricks in Lovelace, but a more deeply implemented behavior would be much more secure than popping a JS box to enter a PIN or passphrase.

The idea is to have an ability to add a property to a `button` entity to make it protected. In a way that, when called either from Lovelace, or via API, it would require a PIN or password be provided, in order to execute the action.

This is a more involved fuature request, of course, as it goes a bit deeper into entities and requires a change also in frontend (to provide a pop-up). But I think it's uhm... kind of due :) at least for a button. I would likely try to provide a pull request if I understood how to code in C...

**Please describe your use case for this integration and alternatives you've tried:**
A real life use case: I have a door with a smart lock (that uses a biometric sensor to open the lock). The door controller has an input, that I can connect to a pushbutton to provide a way to unlock the lock from some other place in the house, without having to go downstairs. So I can place a relay connected to an esphome node, then I can pretty much do it from anywhere. Cool ! That's great.

But, that also means anyone with access to HA UI can simply puch the button and unlock the door. My kid can grab my mobile or tablet and push this button while we're far away from home. For some entities it makes a lot of sense to put a PIN prompt in home assistant. So, I could have button to unlock my main door, or remotely open the gates from 1000 km away -> but I'd fell a lot better knowing that I also need to enter the right PIN when doing that.

Because... Unlocking the house is not the same thing as turning on a light.

Perhaps there is some way to do this already, but I searched and haven't found a good way. I did find some Lovelace-based PIN prompts, but those are not secure enough, they rely on the UI bloking your way in, not the actual PIN -> that could, in the backend, have fail counts, lock-outs, avoiding brute force attacks, etc -> making this quite secure.

If it would only exist for a button, nothing elase, it would be more than great !

**Additional context**

lucius-the, 2023-01-13:
Thank you for the information!
This PR enables input of strings in esphome through some hardware. Which is a great thing to add to esphome, sure. It is related, cover one of the things but... doesn't solve the issue.

Esphome would, in fact, need to accept a PIN as an input parameter for `button.press` action for this to work properly.
The thing is, both HA and esphome (and their underlaying button entity) should first provide an option to require an input parameter for `button.press` and then we would need HA to provide a UI to input that parameter when it's configured as required for a button press in esphome. That's how I see this would probably need to work...

lucius-the, 2023-01-13:
So this needs a change both in esphome and Home Assistant.

lucius-the, 2023-01-13:
At the moment I'm trying a find a workaround, but I'm having problems finding a way to push data from HA to esphome. The idea is I could maybe define a service on esphome and then call it from HA. But that service needs to accept a string somehow (a PIN).
We have a text sensor, but AFAIK this works the other way around. I can _send_ text from esphome to HA, but not in reverse... If I could do that, then I could use some lambda on esphome side to validate input and decide weather to take action. Anyone knows of a way I could achieve that ? Sending a string from HA to esphome ?

nagyrobi, 2023-01-13:
`text_input` is for inputting text to ESPHome through HA so likely that's what you want, see linked PR above, work is in progress.

And yes, for proper and easy usage it needs some support added to HA, unless you're hacky.
You might also look at HA's own `text_input`: https://www.home-assistant.io/integrations/input_text/
There's ` pattern` config option for validating with a Regex.

lucius-the, 2023-01-13:
Thank you ! Will look into this.
At the moment I'm looking at https://esphome.io/components/api.html#user-defined-services because it seems I can define a service in esphome that accepts variable input, after all...

Combining that with a text sensor might do the trick. I don't like the text sensor though, because I think it would record state - and I don't want to see the PIN that was entered later in history... or pass it around HA layers unneccesarily. So the service looks a bit better. But then, I still need to figure out a way for a PIN prompt in Lovelace. Something that doesn't get saved anywhere, just passed through a service call...

Well, I feel I'm much closer now ;)

Thank you for good pointers ! Perhaps I should have opened this on Discord or Reddit first, but well.

nagyrobi, 2023-01-13:
> I still think it would be cool if we had a button that accepts a PIN, and also pops up a prompt in HA UI if PIN is required for button.press

That request is 80% HA and 20% ESPHome, so I suggest to rather submit it in HA forums first...

lucius-the, 2023-01-13:
Yes, I guess you're right.

JSubelj, 2024-12-01:
Hi, maybe you already figured it out, maybe not, but for someone that gets on this thread maybe this will help. I created just what you want from [alarm_control_panel](https://esphome.io/components/alarm_control_panel/index.html) in esphome and I forked a custom alarm panel and added costumisation [AlarmPanelMoreCustomisation](https://github.com/JSubelj/AlarmPanelMoreCustomisation/tree/master). 

The state logic:
1. armed_away - normal operation, door closed, when code is inputed, state goes to (2.)
2. disarmed - this is the signal to open the door, arms alarm back, so it can trigger `pending` state (3.)
3. pending - door opens, and `pending_time` starts to count down, then it goes to  `triggered` state (4.)
4. triggered - door closes, alarm is automatically disarmed (don't forget to change the pin), and then back to state (1.)

My esphome yaml:
```
alarm_control_panel:
  - platform: template   
    id: alarm_test
    name: "Alarm Test"
    pending_time: "5s"
    codes:
      - "1234"
    on_triggered:
      then:
        - switch.turn_off: open_door
        - alarm_control_panel.disarm: 
            id: alarm_test
            code: "1234"
        - alarm_control_panel.arm_away: alarm_test
        - lambda: id(door_opening_sequence) = false;
    on_disarmed:
      if: 
        condition:
          lambda: return !id(door_opening_sequence);
        then:
          - alarm_control_panel.arm_away: alarm_test
          - lambda: id(door_opening_sequence) = true;  
          - alarm_control_panel.pending: alarm_test 
    on_pending: 
      then:
        - switch.turn_on: open_door
```

My card configuration:
```
type: custom:alarm_control_panel-card
entity: alarm_control_panel.alarm_test
states:
  - arm_away
  - arm_home
durations:
  pending: 5
show_countdown_timer: true
color_countdown_timer: red
labels:
  ui.card.alarm_control_panel.disarm: Open
  state.alarm_control_panel.armed_away.icon.text: closed
  state.alarm_control_panel.armed_away.icon.mdi: mdi:lock-check
  state.alarm_control_panel.armed.icon.label_badge_color: label-badge-green
  state.alarm_control_panel.triggered.icon.mdi: mdi:door-open
  state.alarm_control_panel.triggered.icon.label_badge_color: label-badge-red
  state.alarm_control_panel.triggered.icon.text: opened
  state.alarm_control_panel.armed_away: Door closed
  state.alarm_control_panel.triggered: Door opened
```

Hope this helps :)