esphome/feature-requests #2059: SHTCX sensor "on_value" support

gurselt, 2023-01-13:
**Describe the problem you have/What new integration you would like**

I'd like to publish values via mqtt, using on_value trigger.

**Please describe your use case for this integration and alternatives you've tried:**

When I tried to use "on_value" with this sensor I get an error:
[on_value] is an invalid option for [sensor.shtcx].
Apart from that the sensor works just fine, i.e. the reading is available from the API and HA.

**Additional context**

I have another sensor, tmp117, which has on_value works fine

esp8266:
  board: esp01_1m

sensor:
  - platform: shtcx
    temperature:
      name: "Living Room Temperature"
      id: shtc3_temp
    humidity:
      name: "Living Room Humidity"
      id: shtc3_rh
    address: 0x70
    update_interval: 60s

nagyrobi, 2023-01-13:
`on_value` is not specific to certain sensors, it's there for all.
You get that error because you're not using proper indentation. Since shtcx is a multi-sensor device, you need ti put `on_value` at the level of the sub-sensor.

gurselt, 2023-01-13:
> `on_value` is not specific to certain sensors, it's there for all. You get that error because you're not using proper indentation. Since shtcx is a multi-sensor device, you need to put `on_value` at the level of the sub-sensor.

Makes sense!
Yes, it works fine, THX!

closed 2023-01-13 by gurselt