esphome/feature-requests #2062: DSMR - Ability to use two keys and uart assigned sensors

SecSelS, 2023-01-13:
**Describe the problem you have/What new integration you would like**

Hi =) We just opensourced this board https://github.com/SecSelS/SecSi-Energy-Monitor and it has 2x p1/rj12 interfaces for p1 energymeters. The problem is that the telegrams in luxemburg (and possible other countries) are encrypted and you need a DSMR-Key. The problem is, that for every Meter you need a seperate key. As our device has two ports, you can read two meters simultaniously (which is important when you have solar cells, as energy-companies place always two meters in this case). As far as we have seen/tested, you can just use one DSMR-Key per uart correct? Also we're not sure if you can assign the sensors to one uart. So we found it was possible to configure multiple uart (as we're using both with a jumper for the p1 ports -> https://github.com/esphome/issues/issues/521) but no possibility to assign multiple DSMR-Keys for seperate uarts. If our assumptions are correct, we wanted to request to implement the feature of seperate DSMR-Keys per uart, and assign the sensors to seperate uarts. If it is just a config-issue, we would be very glad if you point us to the correct yaml syntax.

**Please describe your use case for this integration and alternatives you've tried:**

The use case would be for households that use our board and have two meters installed. (See above)

**Additional context**

The yaml we use at the moment, the complete Hardware-Docs and design-files are in our github. If you need any more information I would be very happy to assist =)

nagyrobi, 2023-01-13:
All I see on https://shop.secsels.com is a _Terms and Conditions_ page. Product is missing.

nagyrobi, 2023-01-13:
What's the expected price, with shipping to Europe?
(I have 2 meters aswell)

SecSelS, 2023-01-13:
we'll calculate all this in the weekend, but shipping to europe will be available =). we also working on opensource enclosure which will also be sold as well, but not yet on the weekend...

ssieb, 2023-01-14:
I think if you had uncommented the second `dsmr`, you would have gotten a compile error that would have guided you to the solution.

```yaml
sensor:
  - platform: dsmr
    dsmr_id: dsmr_p1_1
    ...
```

SecSelS, 2023-01-14:
Yes that's true for the sensors, but not for the key-section...I was not able to test it on a real meter as in Luxembourg all meters are encrypted...so to connect it to a second meter I need to define a second DSMR key...so it's more about this section: 

```
dsmr:
# here the dsmr for the first p1 port is defined
  id: dsmr_p1_1
  decryption_key: !secret dsmr_key_p1_1
  max_telegram_length: 1700
# here the dsmr for the second p1 port is defined
# use only if you have a second meter connected,
# and the pins are configured for uart-communication
#  - id: dsmr_p1_2
#    decryption_key: !secret dsmr_key_p1_2
#    max_telegram_length: 1700
```

I tried to define a list here (like in the comment and like with uart) but it gives a compile error when I add a second key...

ssieb, 2023-01-14:
Show the config with that part uncommented, because what you have there would be invalid yaml if you uncomment that.

SecSelS, 2023-01-14:
I tried it like in the uart section as list:

```
uart:
  - rx_pin:
      number: GPIO16
    baud_rate: 115200
    rx_buffer_size: 1700
    id: uart_p1_1
# Here the second P1 UART is defined (just uncomment if you use second p1)
  - rx_pin:
      number: GPIO3
    baud_rate: 115200
    rx_buffer_size: 1700
    id: uart_p1_2

# here the dsmr for the first p1 port is defined
dsmr:
  - uart_id: uart_p1_1
    id: dsmr_p1_1_key
    decryption_key: !secret dsmr_key_p1_1
    max_telegram_length: 1700
# here the dsmr for the second p1 port is defined
# use only if you have a second meter connected,
# and the pins are configured for uart-communication
  - uart_id: uart_p1_2
    id: dsmr_p1_2_key
    decryption_key: !secret dsmr_key_p1_1
    max_telegram_length: 1700
 
sensor:
  - platform: dsmr
    dsmr_id: dsmr_p1_1
    energy_delivered_lux:
      name: "Energy Consumed"
      state_class: total_increasing
```

But when I try to compile I get: "Failed config, expected a dictionary" for the DSMR secretion...

ssieb, 2023-01-14:
Ok, now that's the real problem.  Try adding this to your config and see if both will work:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: dsmr
    components: [ dsmr ]
```

SecSelS, 2023-01-14:
This compiles perfectly and reading looks good (couldn't test with a second meter here, but should be ok) =) great great thanks! =) will this be merged? Or do I always need the external component?

SecSelS, 2023-01-14:
Here is my working Test-configuration for reference...I striped away the comments to make it better readable.

```
substitutions:
  device_name: secsi-energy-monitor
  device_description: Energy meter by secsels.com
  ap_fallback_password: secsi-device
     
esphome:
  name: ${device_name}
  name_add_mac_suffix: false
  comment: "${device_description}"

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  ap:
    ssid: ${device_name}
    password: ${ap_fallback_password}
    ap_timeout: 15s
 
captive_portal:
 
logger:
  baud_rate: 0

api:

ota:
  safe_mode: true

web_server:
  port: 80

uart:
  - rx_pin:
      number: GPIO16
    baud_rate: 115200
    rx_buffer_size: 1700
    id: uart_p1_1

  - rx_pin:
      number: GPIO3
    baud_rate: 115200
    rx_buffer_size: 1700
    id: uart_p1_2

dsmr:
  - uart_id: uart_p1_1
    id: dsmr_p1_1_key
    decryption_key: !secret dsmr_key_p1_1
    max_telegram_length: 1700

  - uart_id: uart_p1_2
    id: dsmr_p1_2_key
    decryption_key: !secret dsmr_key_p1_1
    max_telegram_length: 1700
 
sensor:
  - platform: dsmr
    dsmr_id: dsmr_p1_1_key
    energy_delivered_lux:
      name: "Energy Consumed"
      state_class: total_increasing
  - platform: dsmr
    dsmr_id: dsmr_p1_2_key
    energy_delivered_lux:
      name: "Energy Consumed2"
      state_class: total_increasing

  - platform: uptime
    name: "${device_name} Uptime"
  - platform: wifi_signal
    name: "${device_name} Wi-Fi Signal"
    update_interval: 60s
 
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: dsmr
    components: [ dsmr ]
```

ssieb, 2023-01-14:
It would be good if you could test it with two meters at the same time.  I didn't look too closely, so I don't know if it was intentionally only for one, but probably not.  If you can't test it, we'll just merge it anyway in a few days.

SecSelS, 2023-01-15:
Okay... I'll try to find someone here with two meters. The way I'll tested was: I made both definitions like above, and then I simply switched ports. Both decrypted well,.and the values were displayed under the right sensor.

nagyrobi, 2023-01-15:
...run a long RJ11 cable to your neighbor...

SecSelS, 2023-01-15:
🤣 
We use rj12 to get power too ;-)

closed 2023-01-16 by jesserockz

SecSelS, 2023-01-16:
Great great thanks for merging :-) I was not able to test with two meters at the same time, but I configured "as there were" two at the same time, and just switched ports (which works perfectly :-)). So I guess its perfect :-). 

@nagyrobi and everyone interested in a board please contact us directly as I don't want to spam here: https://secsels.com/kontakt/

SecSelS, 2023-01-23:
Confirmed working on two meters at the same time =)

SecSelS, 2023-04-06:
> All I see on https://shop.secsels.com is a _Terms and Conditions_ page. Product is missing.

Just to notify @nagyrobi we're in stock =)