esphome/feature-requests #2065: IR Remote Climate: Panasonic inverter

bengtdg, 2023-01-14:
**Describe the problem you have/What new integration you would like**

Would like  a panasonic inverter to be added til compatible units list. 
The specific model i have in mind is
Panaosonic AIR CONDITIONER R410A
MODEL NO: CS-E12GFEW 
PRODUCED: 2007,04

**Please describe your use case for this integration and alternatives you've tried:**

The mentioned AIR CONDITIONER unit is "Missing" from the Compatible units list

**Additional context**

codyc1515, 2023-12-14:
Same here but for CS-E15PKR

Mrsash2020, 2024-07-01:
Same but for Panasonic CS-E9NKR

Santos-m, 2025-07-05:
Same for Panasonic cs bz35zke currently I'm using the heatpumppir but it's not compatible with remote_receiver