esphome/feature-requests #2067: Add SPI support ADE7953 (used in Shelly Pro 4PM)

phiten, 2023-01-14:
**Describe the problem you have/What new integration you would like**

Add SPI support for the ADE7953 used for power monitoring in Shelly Pro devices. Shelly Pro 4PM uses two of them on one bus.

**Please describe your use case for this integration and alternatives you've tried:**

Add power metering for Shelly Pro devices.

**Additional context**

angelnu, 2023-09-29:
In order to accelerate getting https://github.com/esphome/esphome/pull/5439 merged it would help if others could validate that it works for them.

So if you use the ADE7953 sensor used in the Shelly Pro series, please give it a try by adding

```yaml
external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  - source:
      type: git
      url: https://github.com/angelnu/esphome-1
      ref: ade7953_split
```

to you yaml. Please also provide feedback if you test with I2C where I also added support for additional metrics reported by the sensor (apparent power, reactive power and power factor).

mwetterw, 2025-01-16:
@angelnu 
I've received a Shelly Pro 2PM.
I got everything working on ESPHome (LEDs, Ethernet, Relays, temperatures, switches, button).
The only thing I'm missing is the ADE7953.

Do you know if the current **ade7953_spi** driver in ESPHome is already compatible?
I've tried to use it for one of the two ADE7953 but it doesn't seem to work (I'm using ESPHome 2024.12.2).
I think the issue may be related to the chip select pins.

I got [following information](https://templates.blakadder.com/shelly_pro_2pm.html) (taken from Tasmota):

```
ADE7953 CS1: GPIO0
ADE7953 RST: GPIO2
ADE7953 CS2: GPIO15

SPI CS: GPIO4
SPI MISO: GPIO12
SPIO MOSI: GPIO13
SPI CLK: GPIO14
```

Not sure how to use the SPI CS, and the ADE7953 CS1 and CS2 to talk properly to the two ADE7953.
I managed to talk to the SN74HC595 through SPI by using SPI CS.

This message from Tasmota maintainer is also very interesting about the CS pins:
https://github.com/arendst/Tasmota/discussions/16733#discussioncomment-3924511

Here is what I've tried (but it doesn't work):

```
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12
sensor:
  - platform: ade7953_spi
    cs_pin: GPIO0
    voltage:
      name: "Voltage"
    frequency:
      name: "Frequency"
    current_a:
      name: "Current A"
    current_b:
      name: "Current B"
    power_factor_a:
      name: "Power Factor A"
    power_factor_b:
      name: "Power Factor B"
    apparent_power_a:
      name: "Apparent Power A"
    apparent_power_b:
      name: "Apparent Power B"
    active_power_a:
      name: "Power A"
    active_power_b:
      name: "Power B"
    reactive_power_a:
      name: "Reactive Power A"
    reactive_power_b:
      name: "Reactive Power B"
```

It also seems I have no way in the ESPHome configuration to specify the reset pin for the ADE7953 (GPIO2).

mwetterw, 2025-01-23:
@angelnu FYI, I'm adding support for the Shelly Pro 2 PM: https://github.com/esphome/feature-requests/issues/3033
Your driver works for it! Thank you. :)

But I faced two issues that remain (listed in the issue). I'm able to get around them just by using YAML configuration work arounds, but it would be better to handle it in the source code.