esphome/feature-requests #2071: Media player: Support media metadata

sgofferj, 2023-01-17:
**Describe the problem you have/What new integration you would like**

MP3 (and other) streams contain metadata, Home Assistant allows setting metadata in the media_player.play_media service.
Currently, ESPHome does not seem to support any metadata in the mediaplayer component. At least there is nothing in the docs and media player cards in HA only show play/pause status but no metadata.
I'd like to suggest adding support for metadata to the mediaplayer component so that e.g. displays can be used on a player to show what's playing and media player cards in HA also show it.

**Please describe your use case for this integration and alternatives you've tried:**

There are no alternatives short of not using ESPHome for media players.

**Additional context**

N/a

reinhapa, 2024-06-18:
Would be great to be able to build a HA connected internet radio though :-)

sgofferj, 2024-06-18:
> Would be great to be able to build a HA connected internet radio though :-)

That's even available as a ready product... Explicitly ESPHome compatible Internet radio...

reinhapa, 2024-06-18:
> That's even available as a ready product... Explicitly ESPHome compatible Internet radio...

I would like to convert an existing Radio Body enabling it to display the current playing titles myself if possible, but do you have any link for what you mean?

sgofferj, 2024-06-18:
Raspiaudio ESPmuse series. IIRC there's also a board only version for projects like yours.

reinhapa, 2024-06-22:
@sgofferj still the ready built product seem to to display the metadata of the currently played internet radio channel as referred in this issue right?

sgofferj, 2024-06-23:
> @sgofferj still the ready built product seem to to display the metadata of the currently played internet radio channel as referred in this issue right?

Yeah, with their own firmware. Not with ESPHome.

reinhapa, 2024-06-24:
> Yeah, with their own firmware. Not with ESPHome.

Would be nice to be able to show the metadata in ESPHome too...