esphome/feature-requests #2072: BME680 BSEC - ESP32-C3 support

Schluggi, 2023-01-19:
The current `bme680_bsec`module does not work with the ESP32-C3.

As i saw recently [here](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME-680-BSEC-for-ESP32-C3/td-p/54638/page/2) there is an inoffical BESC version that should work with the ESP32-C3 as well. 

Maybe someone talented (like @trvrnrth) could implement this? 🥺
(This would help us a lot with our project [AIOsense](https://github.com/Schluggi/AIOsense/tree/v2.0)!)

branhill, 2023-01-22:
Bosch released [BSEC2](https://github.com/boschsensortec/Bosch-BSEC2-Library/releases/tag/v1.3.2200) with ESP32-C3 supports, use the fork from  https://github.com/esphome/feature-requests/issues/1332#issuecomment-1040484949 with latest `BSEC2 Software Library` version `1.4.2200` in [__init__.py#L76](https://github.com/neffs/esphome/blob/9f428916a7d7fb9e88d3d6a994b879ce8098c3d0/esphome/components/bme680_bsec/__init__.py#L76) works for me.

[image]

Schluggi, 2023-01-22:
Thank you @branhill. I'll try it out :)

Schluggi, 2023-01-28:
@branhill This does not work for me.

I forked his repo and changed the bsec version: https://github.com/schluggi/esphome/blob/bsec2/esphome/components/bme680_bsec/__init__.py

Than I extended my yaml:
```yaml
external_components:
  - source: github://Schluggi/esphome@bsec2
    components:
      - bme680_bsec
```

But compiling (even after cleanup build files) failed:
```
Compiling .pioenvs/aiosense-bedroom/src/esphome/components/esp32/preferences.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
src/esphome/components/bme680_bsec/bme680_bsec.h:12:10: fatal error: bsec2.h: No such file or directory

***************************************************************
* Looking for bsec2.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec2.h"
* Web  > https://registry.platformio.org/search?q=header:bsec2.h
*
***************************************************************

 #include <bsec2.h>
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/aiosense-bedroom/src/esphome/components/bme680_bsec/bme680_bsec.o] Error 1
========================= [FAILED] Took 19.32 seconds =========================
```

Any ideas whats wrong?

branhill, 2023-01-29:
Did you change the esp32 framework to `esp-idf` in your yaml? The BSEC 2 library seems only supports `arduino` ([bsec2.h#L45](https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/aeb4e4931fccdd8524c0b40d14aa2c5eec599ff6/src/bsec2.h#L45)).

Schluggi, 2023-01-30:
> Did you change the esp32 framework to `esp-idf` in your yaml? The BSEC 2 library seems only supports `arduino` ([bsec2.h#L45](https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/aeb4e4931fccdd8524c0b40d14aa2c5eec599ff6/src/bsec2.h#L45)).

Yes I had. Thank you. But it does not work anyway:
```
Compiling .pioenvs/aiosense-bedroom/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.h:12,
                 from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
.piolibdeps/aiosense-bedroom/BSEC2 Software Library/src/bsec2.h:50:10: fatal error: bme68xLibrary.h: No such file or directory

***********************************************************************
* Looking for bme68xLibrary.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bme68xLibrary.h"
* Web  > https://registry.platformio.org/search?q=header:bme68xLibrary.h
*
***********************************************************************

 #include "bme68xLibrary.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aiosense-bedroom/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================= [FAILED] Took 28.55 seconds =========================
```

Maybe you can share your yaml?

Schluggi, 2023-01-30:
I close this issue because of #1332. Thanks again to @branhill.

**;TLDR (for all other users)**
Add this to your yaml (only the `arduino` framework is supported):
```yaml
external_components:
  - source: github://Schluggi/esphome@bsec2
    components:
      - bme680_bsec
```

closed 2023-01-30 by Schluggi