esphome/feature-requests #2075: Enable / Disable script function

tiimsvk, 2023-01-24:
**Describe the problem you have/What new integration you would like**
Would it be possible to add a function to the script to enable/disable?

something like:
```
script.enable
script.disable
```

How is it possible in the home assistant with automations and scripts.

It would help especially when filtering the execution of various scripts

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2023-01-24:
You could add a template switch as a condition at the beginning of the switch, so it executes the actions only when it's on.

tiimsvk, 2023-01-24:
yes, I use this option, but it's about adding another sensor and another condition