esphome/feature-requests #2079: Could https://web.esphome.io/ be made to create a generic firmware with a generated hostname, ready for homeassistant 'adoption'?

marcosscriven, 2023-01-25:
**Describe the problem you have/What new integration you would like**
I recently used https://web.esphome.io/  to flash a generic ESP32 USB board, which worked great. I was really impressed how it then magically appeared in ESPHome web gui ready for adoption, without first creating a yaml file.

However, this doesn't work with something like Tuya-convert, which first needs you to flash the firmware first - and that means creating a device/yaml with a set hostname. If you're flashing lots of devices, it would be great to just get them into an 'adoption' state, with a generic hostname you can set at the time of adoption.

**Please describe your use case for this integration and alternatives you've tried:**

Creating a yaml first and then uploading the firmware works fine. It's just a pain when flashing a dozen plugs.

**Additional context**

Not really - except to say thank you. Amazing how far this project has come!

clinta, 2023-03-06:
This is already possible. Create a generic yaml with the option:

```yaml
esphome:
  name_add_mac_suffix: true
```

You can flash this to any device to have it ready for adoption.

clinta, 2023-03-06:
Yes. The only downside is your generic yaml with this option will show up in the esphome dashboard as an offline device. But once a device is flashed with it, it will pop up as a new device ready to adopt.

marcosscriven, 2023-03-06:
In that case it is a little bit different (I think). 

Somehow (I don’t know how), it you go to https://web.esphome.io/, you can flash a generic firmware to your device. 

The in the ESPhome dashboard it’ll appear as online and you just click adopt. I’m really curious how it does that.