esphome/feature-requests #2082: pipsolar: implement command QPGS0

gianfrdp, 2023-01-26:
**Describe the problem you have/What new integration you would like**
pipsoloar component reads just 1st MPPT for inverters having 2 MPPT

**Please describe your use case for this integration and alternatives you've tried:**
I have a Voltronic Axpert MAX 7.2kW inverter with 2 PV input (2 independent MPPT).
Accordly to protocol version 30 for MAX inverters (see https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI30MAX.Communication.Protocol20210217.pdf), QPIGS reads just 1st MPPT.
Using QPGS0, return both MPPT data.

Trying mpp-solar:

```
mpp-solar --port /dev/hidraw2 --protocol PI30MAX --command QPGS0
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
parallel_instance_number        Not valid
serial_number                   92932210103877
work_mode                       Standby Mode
fault_code                      No fault
grid_voltage                    213.4           V
grid_frequency                  49.98           Hz
ac_output_voltage               0.0             V
ac_output_frequency             0.0             Hz
ac_output_apparent_power        0               VA
ac_output_active_power          0               W
load_percentage                 0               %
battery_voltage                 1.1             V
battery_charging_current        0               A
battery_capacity                0               %
pv1_input_voltage               0.0             V
total_charging_current          0               A
total_ac_output_apparent_power  0               VA
total_output_active_power       0               W
total_ac_output_percentage      0               %
is_scc_ok                       0               bool
is_ac_charging                  0               bool
is_scc_charging                 0               bool
is_battery_over_voltage         0               bool
is_battery_under_voltage        2               bool
is_line_lost                    0               bool
is_load_on                      0               bool
is_configuration_changed        0               bool
output_mode                     single machine
charger_source_priority         Solar only
max_charger_current             60              A
max_charger_range               120             A
max_ac_charger_current          2               A
pv1_input_current**               0               A
battery_discharge_current       0               A
pv2_input_voltage               0.0             V
pv2_input_current               0               A
```

**Additional context**

gianfrdp, 2023-01-27:
I found https://github.com/syssi/esphome-pipsolar/tree/pip8048
This meets my needs

JNB89, 2023-09-16:
Hi @gianfrdp, I am a complete noob, but I want to learn. I have a MX2 Voltronic, with the round screen. I Cant get any data from the inverter. My log ends with:
`[22:32:27][C][web_server:161]: Web Server:
[22:32:27][C][web_server:162]:   Address: esphome-web.local:80
[22:32:27][C][mdns:112]: mDNS:
[22:32:27][C][mdns:113]:   Hostname: esphome-web-0244a8
[22:32:27][C][ota:093]: Over-The-Air Updates:
[22:32:27][C][ota:094]:   Address: esphome-web.local:3232
[22:32:27][C][api:138]: API Server:
[22:32:27][C][api:139]:   Address: esphome-web.local:6053
[22:32:27][C][api:143]:   Using noise encryption: NO
[22:32:27][C][improv_serial:032]: Improv Serial:`

Any advice will be welcome! I am sure I am missing something stupid, but to be honest I am not sure I've even copied the code correctly.