esphome/feature-requests #2084: Add the 1 euro filter

MarkoPaasila, 2023-01-28:
**Describe the problem you have/What new integration you would like**
I would like the [1 euro filter](https://gery.casiez.net/1euro/) to be an option.

**Please describe your use case for this integration and alternatives you've tried:**
This is a simple filter which - compared to existing filters - smooths well AND reacts quickly.

**Additional context**
The linked page lists implementations in programmin languages, and also links to a demo where one can test the 1 euro filter and others - and compare them.

nagyrobi, 2023-02-05:
I wish they didn't name it so stupidly.

MarkoPaasila, 2024-05-20:
Arduino version: https://github.com/casiez/OneEuroFilterArduino

ssieb, 2024-05-21:
What would it be used for?  What are you filtering?

netweaver1970, 2024-05-22:
very interesting. Might be nice to collect standard tuning values, say room temperature readings, power monitoring readings etc .... now I'm getting from the sensors a (too) noisy signal, while I know the values can be more steady, as slowly changing.

Ideally it would be an HA option on a sensor, to apply a filter of choice with the required parameters and some sensible defaults (a bit like the decimal precision setting), already in the HA interface, BEFORE saving into the DB, as they are delivered from the -say zigbee- devices. 
But that would require a standard sensor attribute behaviour change in HA. Probably not going to happen anytime soon. 

I guess the only thing one can easier do is then put it indeed in ESPHOME (for custom sensors). 
For standard sensors, one might create template/filter sensors. But that's a maintenance nightmare, as they'll then all exist 2x, once as noisy sensor and once as filtered sensor.
Keeping an eye on this, hopefully someone can do the programming

MarkoPaasila, 2024-05-22:
> What would it be used for? What are you filtering?

I would use it for everything; at least power, temperature, humidity. Most often when you filter enough to get rid of noise, the lag becomes significant. This filter simply seems like a better filter - which would hopefully result in more accurate data while using less storage, in more responsive and accurate automations/logic. I don’t know if it would make any practical difference, but that’s the case with most incremental improvements anyway.