esphome/feature-requests #2085: Add binary sensor option to generic output switch

tradeJmark, 2023-01-30:
**Describe the problem you have/What new integration you would like**

To quote from the source code comments about Switch components, "A switch is basically just a combination of a binary sensor (for reporting switch values) and a write_state method that writes a state to the hardware." Since an Output is more or less a wrapper around a `write_state` sort of method, it's fair to say that a switch is essentially a combination of a binary sensor and an output. However, currently there is no switch platform that will combine an existing binary sensor with an output. The closest is the Generic Output Switch, which will use an Output to write its state, but it doesn't take a binary sensor to confirm the state of the switch, it operates in optimistic mode. I propose adding an optional binary sensor argument to the generic output switch that would take it out of optimistic mode and use the sensor to determine state.

**Please describe your use case for this integration and alternatives you've tried:**
You can already implement this as a template switch with boilerplate that goes roughly like this:
```yaml
switch:
  platform: template
  name: "a switch for the example"
  id: my_switch
  optimistic: False
  turn_on_action:
    output.turn_on: an_output
  turn_off_action:
    output.turn_off: an_output
  lambda: return id(a_binary_sensor).state;
```
It's just a bit hacky and unclean. It would be much nicer and more user-friendly to be able to provide the generic output platform with a binary sensor, and then the config would collapse down to this:
```yaml
switch:
  platform: output
  name: "a cleaner switch"
  id: clean_switch
  output: an_output
  status_sensor: a_binary_sensor
```
`status_sensor` as a name is up for debate, maybe it's just as good to name it `binary_sensor`, although I'd probably prefer the former name as being a little more descriptive.

**Additional context**

I'd be perfectly happy to take a crack at implementing this myself, just want to post it here first to make sure people are onboard and to solicit opinions about the name of the argument.

ssieb, 2023-01-31:
I don't understand the use case here.  What is an actual scenario where you would need this?

tradeJmark, 2023-01-31:
That's approximately the same question as "why would you ever not want to be in optimistic mode", isn't it? If the output's performing any complex action that has a chance of failure, you wouldn't want to assume it got performed, you'd want verified status if that was available. I guess what I'm looking for more than anything is a way to take the generic output platform out of optimistic mode, and I assume there's no objection to that part, at least. As for the mechanism, you could go the same route as the template platform and add in a lambda to check status, but to me you'd almost always model the status check as a binary sensor anyway and then just put in a boilerplate lambda like the one above (that's even the example used on the template documentation page, just a status check on an existing binary sensor), so I think the most user-friendly method for letting people leverage existing binary sensors is to just specify one directly. If you need something more complex, you can always roll your own template binary sensor anyway and that works out the same as if there was a lambda here.

ssieb, 2023-01-31:
What kind of `output` has a complex action with a chance of failure?
I really don't see the point here.  This seems like a very unusual use case that can be solved with the existing automation instead of complicating simple components.

tradeJmark, 2023-01-31:
Something like the BLE Client could, and definitely some forms of template output.

tradeJmark, 2023-01-31:
As an example: Maybe you have a template output that spins a stepper the rough amount to open some blinds, and you have a magnet glued onto the end of the blinds that interacts with a hall effect probe at the top, you could use the hall sensor and make it binary with the threshold platform, and then the blinds open/close switch becomes a pretty simple combination of that output and that sensor.

nagyrobi, 2023-02-05:
I agree with @tradeJmark that indeed, there are many cases when we need to be 100% sure that the switch has indeed turned on, because it's not just a GPIO set high, there can be a lot of complexity between the ESP and the actual outcome which would need a proper feedback to show the _real_ state and nothing failed in between.
This is kinda the thought behind https://github.com/esphome/feature-requests/issues/1568.

But I also agree with @ssieb that the cases can be very compex and they don't limit to generic output switch (I have the same need for a ModBUS switch), so I also think that the power of ESPHome is exactly that most of these can be covered with template switches, just like described in the first post - and that should be enough.

closed 2023-02-05 by nagyrobi

tradeJmark, 2023-02-05:
I guess my thing here is that it seems like we agree there's an upside to doing it, and I'm struggling to understand the downside to doing it. The usage doesn't change on the generic output for people who want to keep using it the same way, it just adds an extra option to make it more convenient in certain situations. We can even document the new option as "advanced" so people don't feel like they need to worry about undstanding it if they don't need it. I'm offering to implement it, so it doesn't take time resources away from the dev team except to take a look at the PR at some point. I don't know, this just seems to me like such an obvious thing to exist - a switch is the combination of an output and a binary sensor, and yet there's no straightforward way to build a switch out of an output and a binary sensor. I don't view this as adding in complexity, I think of it as plugging a gap in the existing functionality - I think it's actually more confusing as a user to discover that there's no way to use the generic output switch except in optimistic mode. That's why I want to put in this one last plug for it. Say the word and I won't bring it up again - I don't want to be annoying - but I just want to make sure I've fully outlined my case.

nagyrobi, 2023-02-05:
The same logic would apply to most other switch integrations too...

tradeJmark, 2023-02-05:
No arguments here. I think any switch component should offer this option unless it's already doing an internal state verification or there's some other reason why it wouldn't apply for that kind of switch. I just find the fact it's missing from generic output switches to be the most glaring, because it's already a switch built out of existing components, so it seemed to me like the logical place to start. But if the only option is to do it for all switches or none, then yeah, maybe that's more trouble than it's worth. Maybe the solution is to make an internal subclass of `Switch` that implements the functionality, which switch implementers can then choose to use or not, sort of like `PollingComponent` vs `Component`.