esphome/feature-requests #2087: Add time calibration for cover based on time

pepsonEL, 2023-01-30:
Hi.  Move my covers to ESPHome from code based on Arduino. In Arduino i setup time for working up and time for working down and also time added name Calibration. In winter and in summer covers works with diffrent speed and time calibration example 3 sec is to correct open or close cover according to weather and temperature.
Now in ESPHome this function is no available. Good will be if this function will be add to ESPHome.

pepsonEL, 2023-01-30:
this is codde which i used on Arduino:
[https://github.com/gryzli133/RollerShutterSplit/]

klaudiusz223, 2023-02-04:
HI @pepsonEL 
I just recently implemented the function you are talking about. I did it with venetian blinds in mind but if you omit the tilt configuration it can be applied to other types of blinds.
Here is the documentation https://github.com/klaudiusz223/esphome-docs/blob/tilt-cover/components/cover/time_based_tilt.rst

```
external_components:
  - source: github://klaudiusz223/esphome@tilt-cover
    components: [ time_based_tilt ]

cover:
  - id: cover_01
    platform: time_based_tilt
    name: "Entrance Cover"
    open_duration: 58700 ms
    close_duration: 57700 ms
    recalibration_time: 2500ms
    close_action:
      - switch.turn_on: blind_01_relay_down
    open_action:
      - switch.turn_on: blind_01_relay_up
    stop_action:
      - switch.turn_off: blind_01_relay_up
      - switch.turn_off: blind_01_relay_down
```

krystof-k, 2023-02-04:
@klaudiusz223 wow, that looks great. Will you create a PR so that it can be merged in ESPHome?

klaudiusz223, 2023-02-06:
@krystof-k 
That's my plan but I want to make sure that the code is stable

krystof-k, 2023-02-07:
@klaudiusz223 cool, happy to beta test it! I'll post further feedback.

klaudiusz223, 2023-02-07:
@krystof-k This is great.
Which feature is the most important to you that is not in the original implementation? e.g. tilt, recalibration, inertia, interlocking?
Which curtains do you want to use this for? e.g. with or w/o tilt?
Do not forget to use the latest version. I've fixed one small issue yesterday.

krystof-k, 2023-02-07:
The tilt. Which seems to be working (within the limits of the blinds with tilt) great.

I'm not sure how exactly the interlock should work, but I had to do this in order to get rid of the relay board "blackouts" when changing direction:

```
    open_action:
      - switch.turn_off: open
      - delay: 10ms
      - switch.turn_on: close
    close_action:
      - switch.turn_off: open
      - delay: 10ms
      - switch.turn_on: close
```

klaudiusz223, 2023-02-07:
I don't quite understand what problem you want to eliminate by interlocking
In my opinion there is no need to set a delay in open and close action. I suspect that this disturb the calculation of the travel time of the blind. The same can be achieved by setting the `interlock_wait_time`. 

This is my configuration together with the switch part (also with interlocking)

```
external_components:
  - source: github://klaudiusz223/esphome@tilt-cover
    components: [ time_based_tilt ]

switch:
  - platform: gpio
    id: "blind_01_relay_down"
    interlock: [blind_01_relay_up]
    disabled_by_default: true
    pin:
      pcf8574: output_exp_01
      number: 0
      mode:
        output: true
      inverted: true

  - platform: gpio
    id: "blind_01_relay_up"
    interlock: [blind_01_relay_down]
    disabled_by_default: true
    pin:
      pcf8574: output_exp_01
      number: 1
      mode:
        output: true
      inverted: true
	  

cover:
  - id: cover_01
    platform: time_based_tilt
    name: "Entrance Cover"
    open_duration: 58700 ms
    close_duration: 57700 ms
    inertia_open_time: 300 ms
    tilt_open_duration: 930 ms
    inertia_close_time: 250 ms
    tilt_close_duration: 900 ms
    interlock_wait_time: 500ms
    recalibration_time: 2500ms
    close_action:
      - switch.turn_on: blind_01_relay_down
    open_action:
      - switch.turn_on: blind_01_relay_up
    stop_action:
      - switch.turn_off: blind_01_relay_up
      - switch.turn_off: blind_01_relay_down	 
```

In my case the interlock time is 500ms because this is what the manufacturer of the blind motor recommends. I set this time even though I have the interlock solved in hardware. In total, interlock is provided in 3 places (1 hardware + 2 software).
As you can see I only switch off the relays in stop_action. 
If you are concerned, you can leave switching off the relays  in the configuration at close_action and open_action but remove the delay

krystof-k, 2023-02-10:
Thanks, I'll further evaluate it and get back. I need to sort out the inrush current which breaks the relay boards first.

dkrebs, 2023-08-07:
@klaudiusz223 Thank you for the contribution! I started using the `tilt-cover` for the threeway valve of my heating. Especially the `recalibration_time` and `interlock_wait_time` are important compared to the original cover as the normal `time-based cover` is loosing its correct position after a while. I will keep you posted if anything goes wrong :-)