esphome/feature-requests #2089: [Ethernet Component] Add JLSEMI_JL11x1 PHY Support

ESP4Ever, 2023-01-30:
**Describe the problem you have/What new integration you would like**

Could you please add a JLSEMI_JL11x1 PHY support to the Ethernet Component?
It announced as a "compatible with RTL8201F only ID different"
Datasheet: [https://github.com/dtlzp/relay_dev_demo/blob/main/datasheet/PHY_JL1101_compatible_RTL8201F.pdf](url)
Driver: [https://github.com/dtlzp/relay_dev_demo/blob/main/esp-idf-dt-patch/components/esp_eth/src/esp_eth_phy_jl1101.c](url)
GPIO mapping:
[image]

**Please describe your use case for this integration and alternatives you've tried:**

I'm trying to develop  a ESPHome-based custom FW for a 8x relay board but it uses a "nobody know" PHY and I/O shift registers whis sharing same ESP32 GPIO (will create another RFF)

Tried this but ESP is fast blinking by status led and doesn't boot

ethernet:
  type: RTL8201
  mdc_pin: 23
  mdio_pin: 18
  clk_mode: GPIO17_OUT
  power_pin: 0
  phy_addr: 1

**Additional context**

nagyrobi, 2023-02-05:
Please post links to the board itself.

nagyrobi, 2023-02-05:
Also see work in progress: https://github.com/esphome/esphome/pull/4009

ESP4Ever, 2023-02-07:
Hello!

Thank you for your reply.
All information available for this board series is here: [Dingtian Relay Development Board](https://github.com/dtlzp/relay_dev_demo)

nagyrobi, 2023-02-07:
It has been merged to the `dev`. Please test.

ESP4Ever, 2023-02-09:
It's unusual - everything worked on the first try, thank you so much!
The only problem I had is to find a new PHY definition in the code)))

Setup:
=====

ethernet:
  type: JL1101
  mdc_pin: 23
  mdio_pin: 18
  clk_mode: GPIO17_OUT
  power_pin: 0
  phy_addr: 1

Result:
=====

[10:09:24][C][ethernet:167]: Ethernet:
[10:09:24][C][ethernet:285]:   IP Address: 192.168.0.36
[10:09:24][C][ethernet:286]:   Hostname: 'test-board'
[10:09:24][C][ethernet:287]:   Subnet: 255.255.255.0
[10:09:24][C][ethernet:288]:   Gateway: 192.168.0.1
[10:09:24][C][ethernet:293]:   DNS1: 192.168.0.1
[10:09:24][C][ethernet:294]:   DNS2: 192.168.0.2
[10:09:24][C][ethernet:301]:   MAC Address: B8:D6:1B:3A:8E:95
[10:09:24][C][ethernet:306]:   Is Full Duplex: YES
[10:09:24][C][ethernet:311]:   Link Speed: 100
[10:09:24][C][ethernet:170]:   Power Pin: 0
[10:09:24][C][ethernet:172]:   MDC Pin: 23
[10:09:24][C][ethernet:173]:   MDIO Pin: 18
[10:09:24][C][ethernet:174]:   Type: Unknown <--------------- ?????????????????

nagyrobi, 2023-02-09:
Is this just cosmetic or it breaks operation?

nagyrobi, 2023-02-09:
You're not answering the question.
Is the ESP32 operating properly over Ethernet with this chip and driver, apart from that `Type: Unknown` message?

And also, please quote your code/log snippets properly.

ESP4Ever, 2023-02-09:
I'm sorry but I had no time to check because since some time, when ESPHome node added to HA it takes much time to get all entities visible except "Firmware" which appears immediately.
What I can tell right now is that the board is online, I can start logging "Wirelessly" and to install FW "Via the network"

closed 2023-02-13 by ESP4Ever