esphome/feature-requests #2091: Please add Validation, or the information it provides, to the Edit window

DeeBeeKay, 2023-01-31:
**Describe the problem you have/What new integration you would like**
I would like the ability to Validate code without having to save and exit the code editor first.

**Please describe your use case for this integration and alternatives you've tried:**
The editor underlines code that is not correct, but provides no detail as to *why*. Validation tells you *why* the error exists. Sometimes code is wrong because of a change in ESPHome that has not made it into the website yet.

**Additional context**
Saving and installing are both available inside the editor. My need for the information Validation provides is usually when I am in the editor. Saving and closing to Validate is extra steps. Also, I just burned a chunk of time hunting for an error in my Neopixel code, the formatting of which was copied from esphome.io, and worked in previous versions. Solved it by Validating in the end, which told me the new config option I needed (chipset, as it happens).

ssieb, 2023-01-31:
I thought there was hover information on what the problem was.
Describe how you think this would work.  What information from the validation would be useful?

DeeBeeKay, 2023-01-31:
Ah. You see, the difficulty is that I'm not super bright. Admins, please feel free to delete.

In my defence, I upgraded from a much older version of ESPHome, specifically to get down with the Pi Nano. I was totally unaware that the popup existed. I'm more of a keyboard than mouse kind of guy. Comes of starting on Atari 800XLs.

I was running old code because, for my use case, it was stable. Many of my devices are Sonoff products that are installed in places my body doesn't enjoy going if I brick them, so, if it aint broke, you know?

Anyhoo. Thanks for the heads up.

closed 2023-01-31 by DeeBeeKay