esphome/feature-requests #2096: Support template vacuum platform

Rjevski, 2023-02-01:
**Describe the problem you have/What new integration you would like**

I have a vacuum robot that has a Tuya MCU in it. I've swapped the Tuya module with an ESPHome module and would like to use the TuyaMCU integration to control it.

At the moment I can map the vacuum's DPs to Tuya entities in ESPHome however those show up as separate entities in Home Assistant. Ideally I'd like to have a template vacuum integration so it can show up as a single vacuum entity in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2023-02-05:
ESPHome doesn't support any vacuum platform. You should ask this in Home Assistant's feature requests instead, there a template for this kind of use case seems more easily feasible...

Rjevski, 2023-02-11:
> ESPHome doesn't support any vacuum platform

Yes, that's the reason behind the feature request!

> there a template for this kind of use case seems more easily feasible...

I've indeed worked around the issue but it's less than ideal as it requires configuration (in YAML no less) on Home Assistant's side.

Ideally I would like the user experience to be equivalent to a light - you can use any kind of `light` component in ESPHome (even a custom one) and it magically appears as a light in Home Assistant, no extra configuration required.