esphome/feature-requests #2097: Add a documentation to devices in the ESPHome add-on for HA

roberttco, 2023-02-01:
**Describe the problem you have/What new integration you would like**
Each device card in the Add-on UI has links for editing, visiting (if a web server is configured), and look at logs.  It would be handy to be able to add details about the device and access it via the card as well.  Maybe to continue the paradigm of a documentation link that other cards have in the integrations, maybe it would make sense to add this to the action menu, or just a link on the card proper.

**Please describe your use case for this integration and alternatives you've tried:**
Not really any alternatives other than adding comments to the code.  In fact, maybe that is what you would display.  Display the content of a specially delineated block of comments in the code 🤔

Another use case would be to have an auto-populated section that has the devices IP address, device hostname, MAC, etc.  or a direct link to the device information as if you had found it in the Devices & Services list.

**Additional context**
Just wanting to make it easy to differentiate between devices that have similar names and/or get information about the device.

nagyrobi, 2023-02-05:
I use `project` configuration for this puprose...

```
substitutions:
  device_name: ble-erzekelok-proxy
  device_description: "ESP32-POE-ISO kontroller: BTHome érzékelők, RFBridge rádiós vezérlés"

esphome:
  name: ${device_name}
  comment: "${device_description}"
  project:
    name: esphome,robi.${device_name}
    version: ${device_description}
```

Results in:
[image]

roberttco, 2023-02-05:
@nagyrobi Excellent.  Thank you for this.

closed 2023-02-05 by nagyrobi