esphome/feature-requests #2098: Allow ESPHome devices to act as DNS Server

felipecrs, 2023-02-02:
**Describe the problem you have/What new integration you would like**

I have a Huawei AX3 Pro router. It's a great piece of hardware, that unsurprisingly lacks a proper software. OpenWRT is likely to never be available to it.

I use AdGuard Home as a HA add-on, acting as a DNS server, and I have a problem:

When my HA instance goes down, the internet connection of the whole home goes down as well. Like for example, when I'm restarting HA.

But another situation is when I have a power outage, which is frequent where I live. The UPS which holds my HA instance can hold it up for like 15 minutes, but the UPS which holds my Huawei router can hold it up for like 45 minutes. In other words, in a power outage I only get internet connection for the firsts 15 minutes.

I can of course manually go in the Huawei web UI and change it to something like 1.1.1.1 to restore connectivity, but doing that manually every time is cumbersome.

**Please describe your use case for this integration and alternatives you've tried:**

If I could make an ESPHome act as a DNS Server, I'd put it as the DNS address of my router. Inside of ESPHome, it would redirect requests to my AdGuard Home add-on if the HA host is up, or otherwise to 1.1.1.1.

**Additional context**

nagyrobi, 2023-02-05:
Get a bigger UPS...

felipecrs, 2023-02-05:
Yeah. I should do that. :)

jace, 2024-01-25:
I just upgraded Home Assistant from an RPi4 to a 5. Because this required downtime, I had to first change my DHCP DNS settings to not use AdGuard Home, then power cycle wifi to force all clients to reconnect and acquire the working DNS server, then restore HA and confirm everything was up, and then reset DNS and force-reconnect everyone again.

Made me think that I should move DNS to a standalone device that requires no maintenance downtime. An ESP32 device for the low power draw (but no ethernet), or a Pi Zero with USB-Ethernet (though 100 Mbit only), or an older Pi2 I have spare 
(though also 100 Mbit).

There's an ESP32-based adblocker already available, but AdGuard Home has nice extras such as a DNS-over-HTTPS backend. https://github.com/s60sc/ESP32_AdBlocker