esphome/feature-requests #2099: Allow !secret to be used for encryption key

aceindy, 2023-02-02:
**Describe the problem you have/What new integration you would like**
In the past I used to use HA's secret file to store my passwords for all my devices

Now with the new encryption it seems it is not possible to use a secret anymore.
# Enable Home Assistant API
api:
  encryption: !secret api_encryption

results in: 1 of 1 problem
expected a dictionary.

It would be handy to use a key from secrets, so all will use the same ;)

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

randybb, 2023-02-02:
You have it wrong:
```yaml
api:
  encryption:
    key: !secret api_encryption
```

aceindy, 2023-02-02:
@randybb sorry, didn't realize formatting was possible from my phone, corrected the formatting.
However i use the correct formatting on esphome config, and will throw an error 🤔

aceindy, 2023-02-02:
You are right, sorry to have bothererd you ...

closed 2023-02-02 by aceindy