esphome/feature-requests #2100: on_time trigger with template at Time component

sdalee, 2023-02-04:
**Describe the problem you have/What new integration you would like**

Many automations want timed events. The time can be set in the Home Assistant dashboard and the set time would be stored in the ESP flash (EEprom). 
**Please describe your use case for this integration and alternatives you've tried:**

Now the on_time trigger can only get hard-coded time (hours, minutes, seconds, months, cron, etc...), meaning that the time values are fixed in ESP, not changeable by a variable, thus from the Home Assistant Dashboard interface.
**Additional context**

The best solution would be if the on_time trigger could be a template for all configuration variables. 
So, for example, in the case of the irrigation controller I'm developing now, watering would start at the time and on the days set in Home Assistant, even if Home Assistant is not available, and would remain at the set value even in the event of a power failure.

nagyrobi, 2023-02-05:
Use NTP as time source if you're afraid of loosing Home Assistant. Some network devices can provide local NTP service you can use locally (routers, OpenWRT, Synology etc.). You can also use a [GPS](https://esphome.io/components/gps.html) as independent time source if you're really paranoid.

exeljb, 2024-08-25:
I was surprised that these couldn't be templated.  I was trying to setup an easy to set timer for my wife to make changes to from the devices webpage without having to go into home assistant/esphome/edit code/complie every time she wants to change the time turns on and off.   She wants this to operate independently from home assistant, so it can't rely on commands from HA, it needs to be all internal to the device.

My plan was to have number sliders for on hours, on minutes and off hours, off minutes, then template or substitute (ie, $on_hr) those values for the corresponding time based actions.  No luck.

For how smart esphome can do a lot of things, it seems like it's dumb when it comes to doing the dumber things.

raythefourth, 2024-08-26:
Back in April, ESPHome added [datetime component](https://esphome.io/components/datetime/index.html) and [template datetime](https://esphome.io/components/datetime/template.html). 

This allows you to set a time in Home Assistant which is updated and stored locally on the ESP device. 

Check out the sprinkler controller I created [here](https://community.home-assistant.io/t/simple-reliable-sprinkler-controller-with-esphome/628838), which uses this functionality.

I think this issue can be closed.