esphome/feature-requests #2101: add support for weact eink (2.13 maybe 2.9)

alinelena, 2023-02-04:
would be nice to add support for weact eink module... 
https://github.com/WeActStudio/WeActStudio.EpaperModule

I have already played with it and is almost working... with this 

```
  57 spi:                                                                                                                                
  58   id: bus_a                                                                                                                         
  59   clk_pin: GPIO04                                                                                                                   
  60   mosi_pin: GPIO06                                                                                                                  
  61   miso_pin: GPIO05   

  71 display:                                                                                                                            
  72   - platform: waveshare_epaper                                                                                                      
  73     spi_id: bus_a                                                                                                                   
  74     model: "2.13in-ttgo-dke"                                                                                                        
  75     reset_pin: GPIO02                                                                                                               
  76     cs_pin: GPIO07                                                                                                                  
  77     dc_pin: GPIO01                                                                                                                  
  78     busy_pin: GPIO3                                                                                                                 
  79     rotation: 90                                                                                                                    
  80     full_update_every: 60                                                                                                           
  81     update_interval: 120s 
```
on an esp32-c3 also tested on a normal esp32 nodemcu

the only issue I noticed is the bottom row is full of noise.. I traced this to the fact the ram on x somehow is shifted. starts at 0x01 but shall start at 0x0 the size is correct...

alinelena, 2023-02-04:
practically I had to change this

```
1513   this->command(0x44);                                                                                                              
1514   this->data(1);                                                                                                                    
1515   this->data(this->get_width_internal() / 8);
```
to

```
1513   this->command(0x44);                                                                                                              
1514   this->data(0);                                                                                                                    
1515   this->data(this->get_width_internal() -1/ 8); 
```

alinelena, 2023-02-04:
[image]

top is the fixed version... below the original one... ignore the extra line of text

phaedrus42a, 2023-02-23:
I managed to get the WeAct Studio 2.9" 128x296 module working without much effort on a NodeMCU ESP32.
Code:

spi:
  
  mosi_pin: GPIO15  # aka SDA
  clk_pin: GPIO02  #aka SCL
  # miso_pin: GPIO15

display:
  - platform: waveshare_epaper
    model: 2.90inV2
    cs_pin: GPIO16
    dc_pin: GPIO17
    reset_pin: GPIO05
    busy_pin: GPIO18
    rotation: 270
    full_update_every: 30
    update_interval: 5s
    lambda: |-    
      it.print(0, 0, id(font1), "Hello World!");    
      it.strftime(0, 64, id(font1), "%H:%M:%S", id(esptime).now());

[image]

swiss01, 2023-02-25:
@phaedrus42a Fuck! You are Giniels :D 

My display works more or less.
I still have problems that the new data is not displayed cleanly and the old ones are still fading for a while. What can be the reason?

I work with an ESP 8266 mini. I connected them as follows.

```
spi:
  clk_pin: D1
  mosi_pin: D2
    
display:
  - platform: waveshare_epaper
    model: 2.90inV2
    cs_pin: RX
    busy_pin: D5
    reset_pin: D8
    dc_pin: D4
    rotation: 90
    full_update_every: 5
    update_interval: 60s

```
[image]

swiss01, 2023-03-05:
ok found the problem. In other words, I know how to deal with it.
I sent a little the wrong code above. Of course I have an update_interval of one second.

However, the display can refresh at most every three seconds, otherwise there are artifacts.

So I removed the seconds display and set the update_interval to 5s. That's how it works very well.

if anyone finds out how to update every second without artifacts, please report :) 

Code for ESP8266
```
spi:
  clk_pin: D1
  mosi_pin: D2
    
display:
  - platform: waveshare_epaper
    model: 2.90inV2
    cs_pin: RX
    busy_pin: D5
    reset_pin: D8
    dc_pin: D4
    rotation: 90
    full_update_every: 14400
    update_interval: 5s
```

alinelena, 2023-03-05:
looking in weact example and library they used i would say 15s is what you should use.

swiss01, 2023-03-05:
Ah ok. now I can see that on the product page on Aliexpress as well. only the translation into German is very bad.
Black and white has about 3s and the tricolor about 19s

the seconds display flies out and the problem is solved 👍

alinelena, 2023-03-15:
just a note: i got my hands on a 2.9 bw seems with 2.9inV2 accummulates artifacts very fast much faster than 2.13, hence the need of faster full refreshes... I will need to check that the settings are actually fine.

swiss01, 2023-03-15:
I have now connected 3 displays to an ESP32. The main problem is the first time the display refreshes, there may be artifacts for a while before there isn't much data left to refresh. After about 1-2 minutes I no longer have any artefacts on the display and the minute display runs well.
Does it work? Yes. Good? no! but it works ;)

I'm trying to replicate it and adapt it to my needs
https://community.home-assistant.io/t/e-paper-display/138625/33

My Code. 
````
esphome:
  name: esp-wetterstation
  friendly_name: ESP-Wetterstation

esp32:
  board: esp32-devkitlipo
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Wetterstation"
    password: "lP5DXqBkjIfa"

captive_portal:

# Ab hier kopie von psp888

font:
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_85_font
    size: 85  
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_75_font
    size: 75
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_60_font
    size: 60
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_50_font
    size: 50
  - file: 'fonts/Roboto-Light.ttf'
    id: size_12_font
    size: 12
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_15_font
    size: 15
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_14_font
    size: 14    
  - file: 'fonts/Roboto-Medium.ttf'
    id: size_20_font
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
[... 365 lines cut ...]
  - platform: homeassistant
    entity_id: sensor.0xa4c1382ae3bae706_temperature
    id: kitchen
    internal: true
  - platform: homeassistant
    entity_id: sensor.0xa4c138d4da5b73ad_temperature
    id: outside_temp
    internal: true
  - platform: homeassistant
    entity_id: sensor.durchschnitt_temp
    id: inside_temp
    internal: true
  - platform: homeassistant
    entity_id: sensor.0xa4c138d4da5b73ad_humidity
    id: outside_humidity
    internal: true
  - platform: homeassistant
    entity_id: sensor.0xa4c1382ae3bae706_humidity
    id: inside_humidity
    internal: true
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    id: "wifisignal"
    update_interval: 15s    
````

alinelena, 2023-03-17:
@swiss01 I had some go at it
https://github.com/esphome/esphome/pull/4579

is not 100% finished I want to triple check some update modes but works much better than 2.9inV2
if you want to test my version use model  
```
model: "2.90inBS"                                                                                                               
```

swiss01, 2023-03-17:
Oh great.
Unfortunately it works worse than 2.9niV2
when updating, the display fades and then becomes sharp again. in general, there are fewer artifacts, but the update behavior is very annoying.
Have attached a gif.

If you have a new version, I'll be happy to test it.

[image]

alinelena, 2023-03-18:
your update rate is too high. increase it the fading is from the update i will need to check the exact params for update mode.

alinelena, 2023-03-18:
just a note... esphome does not really support partial updates... it supports the mechanics of them... pushing the right instructions but the area which is updated is always the full screen.
so having fast update rates is not healthy for the screen

alinelena, 2023-03-18:
@swiss01 try the latest bc14ef04b8c9bdd7595752ac33581bed54d9e204 in my testing seem to behave properly... please note my note above

alinelena, 2023-03-18:
pull request #4579 addressing the issue

swiss01, 2023-03-19:
I set the update rate to 10s.

I have now tested it extensively. I'm not sure where the error is buried but. with the use of an ESP8266 it works very well. when changing the whole display flickers very little. So the background is slightly darker for a moment.

I have more problems with my ESP32. I've attached 3 displays. The problems are very diverse. Partly I think it's because of the wiring, but then again because of my YAML file and then again because of the library.

The errors that appear.
- Display only shows noise
- Display is mirrored.
- Display only shows two images (time jumps back and forth between 13:45 and 13:46)
- Display only do a full reset and then nothing at all.

For a long time I thought it was down to the planning. Then connect them all in series. Sometimes the last display has problems, sometimes the middle one, sometimes the first.
[image]
[image]
[image]

swiss01, 2023-03-19:
ok now i got the three display to work.
I have divided the displays on two SPI buses. The library works well so far, except for the slight flickering when updating.
But I don't know if this is normal or if there is still a bug in the library.

alinelena, 2023-03-20:
i do not think daisy chaining is supported in esphome, using different cs pins and share the rest shall work. now the update change of intensity at a partial update i do not know if is normal or not I see it myself too... but is not annoying enough to look more into it.

alinelena, 2023-03-20:
for reference this is my example code for 2.9 https://gitlab.com/48broadbent/broadbent/-/blob/develop/jupiter.yaml 
and this for 2.13
https://gitlab.com/48broadbent/broadbent/-/blob/develop/smallbedroom.yaml

swiss01, 2023-03-20:
> i do not think daisy chaining is supported in esphome, using different cs pins and share the rest shall work. now the update change of intensity at a partial update i do not know if is normal or not I see it myself too... but is not annoying enough to look more into it.

I did that too. CS separately on the EPS and the other cables are bridged together. I simply used the two terminals on the Weact as a bridge. I had previously wired it through a breadboard. but wasn't better. and a pure cable salad.
With two SPI it works fine.

swiss01, 2023-04-20:
how long do you think it's going to be in an official version?

deltamelter, 2024-03-23:
Would it be too much to expect support for 3 colour 2.9" ?  Or is this likely to just work once the weact panels are supported?

regystro, 2024-03-23:
> Would it be too much to expect support for 3 colour 2.9" ? Or is this likely to just work once the weact panels are supported?

There's a pull request already working. https://github.com/esphome/esphome/pull/6226
See an example:  https://github.com/esphome/feature-requests/issues/2191#issuecomment-2006760975

deltamelter, 2024-03-23:
> There's a pull request already working. [esphome/esphome#6226](https://github.com/esphome/esphome/pull/6226) See an example: [#2191 (comment)](https://github.com/esphome/feature-requests/issues/2191#issuecomment-2006760975)

That's great news.  what do I need to replace to try this out?  I currently use esphome as a HA plugin.

regystro, 2024-03-24:
> That's great news. what do I need to replace to try this out? I currently use esphome as a HA plugin.

Just add:
```
external_components:
  - source: github://pr#6226
    components: [waveshare_epaper]
```

deltamelter, 2024-03-25:
Thanks but I also need to add SPI, CS, DC, RESET an BUSY PINS?
I tried to add the same pins as worked with arduino sample but it just boot loops.
Any ideas? I threw this together from bit of information from these threads.

```
GxEPD2_DISPLAY_CLASS<GxEPD2_DRIVER_CLASS, MAX_HEIGHT(GxEPD2_DRIVER_CLASS)> display(GxEPD2_DRIVER_CLASS(/*CS=D8*/ EPD_CS, /*DC=D3*/ 0, /*RST=D4*/ 2, /*BUSY=D2*/ 4));
```
```
esp8266:
  board: esp_wroom_02

sensor:
  - platform: uptime
    name: Uptime Sensor  
    id: uptime_text
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13

external_components:
  - source: github://pr#6226
    components: [waveshare_epaper]
font:
  - file: 'Roboto-Regular.ttf'
    id: roboto
    size: 24    
color:
  - id: red
    hex: "FF0000"
  - id: black
    hex: "FFFFFF"
display:
  - id: screen
    platform: waveshare_epaper
    model: 2.90in3c
    cs_pin: GPIO15
    dc_pin: GPIO00
    reset_pin: GPIO02
    reset_duration: 2ms
    busy_pin: GPIO04
    rotation: 90
    update_interval: 300s # 5m    
    lambda: !lambda |-
      it.print(0, 4, id(roboto), id(red), "ESPHome ePaper");
      it.print(220, 1, id(roboto), id(red), "E");
      it.print(235, 0, id(roboto), id(black), "E");
      it.printf(10, 60, id(roboto), "uptime: %d",id(uptime_text).state);
```

regystro, 2024-03-25:
Yep, you'll need those too.
This is my working YAML, but I'm using a different board: esp32dev

```
spi:
  clk_pin: GPIO18   # aka SCL
  mosi_pin: GPIO23  # aka SDA

display:
  - platform: waveshare_epaper
    model: 2.90in3c
    cs_pin: GPIO17
    dc_pin: GPIO22
    reset_pin: GPIO19
    busy_pin: GPIO21
```
I suggest you double check the pins are the right ones based on your board, enable ESPHome debug level and have a look at some ESPHome tutorials and FAQs for troubleshooting.

deltamelter, 2024-03-25:
PIN config looks correct (same as working arduino pin config)
[image]
The log shows the following boot loop exception:
```
[I][logger:034]: Log initialized
[C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[D][spi:039]: Setting up SPI bus...
[D][spi_device:379]: mode 0, data_rate 2000kHz

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40229002 epc2=0000000 epc3=0x00000000 excvaddr=0x3ffe8eba depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd80 end: 3fffffd0 offset: 0190
3fffff10:  3ffe8eb9 3fff2250 3fff21fc 402307c5
3fffff20:  001e8401 00000001 00000000 00000004
3fffff30:  3ffe8eb9 3fff2250 3fff21fc 402173a4
3fffff40:  402172e5 3fff21 3ff21fc 3fff2028
3fffff50:  3fff21fc 00000004 3fff21fc 40217731
3fffff60:  00000000 3fff2328 3fff21fc 4021d03a
3fffff70:  3fff2094 00000003 3fff0224 4021c9d9
3fffff0:  0000000b 0000000b 3fff22cc feefeffe
3fffff90: 00▒▒▒▒2▒▒21c 00000001 feefeffe
3fffffa0:  feefeffe feefeffe feefeffe 3fff062c
3fffffb0:  3fffdad0 00000000 3fff0618 4022ce38
3fffffc0:  feefeffe feefeffe 3ffe8610 40100459
<<<stack<<<

--------------- CTERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007df90
~ld
[I][logger:034]: Log initialized
[C][ota:483]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:490]: Boot loop detected. Proceeding to safe mode.
[E][ota:159]: Error set: unspecified
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
```

Re-compiled the original arduino sample and it still works using the same pins and config.  
```
#define PIN_SPI_SS   (15)     /*CS=D8*/ EPD_CS
#define PIN_SPI_MOSI (13)
#define PIN_SPI_SCK  (14)
/*DC=D3*/ 0
/*RST=D4*/ 2
/*BUSY=D2*/ 4
```

esphome version: 2024.3.0

alinelena, 2024-03-25:
can you remove this.
```
    reset_duration: 2ms
```

deltamelter, 2024-03-25:
Same result, goes straight into boot loop...
```
Showing logs:
[22:04:57][I][logger:034]: Log initialized
[22:04:57][C][ota:483]: There have been 0 suspected unsuccessful boot attempts.
[22:04:57][I][app:029]: Running through setup()...
[22:04:57][D][spi:039]: Setting up SPI bus...
[22:04:57][D][spi_device:379]: mode 0, data_rate 2000kHz
[22:04:58]
[22:04:58]-------------- CUT HERE FOR ECEPTION DECODER ---------------
[22:04:58]
[22:04:58]Exception (9):
[22:04:58]epc1=0x40228ffa epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8eba depc=0x00000000
[22:04:58]
[22:04:58]>>>stack>>>
[22:04:58]
[22:04:58]ctx: cont
[22:04:58]sp: 3ffffd80 end: 3fffffd0 offset: 0190
[22:04:58]3fffff10:  3ffe8eb9 3fff2250 3fff21fc 40207bd  
[22:04:58]3fffff20:  001e8401 00000001 00000000 00000004  
[22:04:58]3fffff30:  3ffe8eb9 3fff2250 3fff21fc 402173a4  
[22:04:58]3fffff40:  402172e5 3fff21fc 3fff21fc 3fff2028  
[22:04:58]3fffff50:  3fff21fc 00000004 3fff21fc 40217731  
[22:04:58]3fffff60:  00000000 3fff2328 3fff21fc 4021d03a  
[22:04:58]3fffff70:  3fff2094 00000003 3fff0224 4021c9d9  
[22:04:58]3fffff80:  0000000b 0000000b 3fff22cc feefeffe  
[22:04:58]3fffff90:  00000014 3fff21fc 00000001 feefeffe  
[22:04:58]3ffffa0:  feefeffe feefeffe feefeffe 3fff062c  
[22:04:58]3fffffb0:  3fffdd0 00000000 3fff0618 4022ce30  
[22:04:58]3ffffc0:  feefeffe feefeffe 3ffe8610 40100459  
[22:04:58]<<<stack<<<
[22:04:58]
```

alinelena, 2024-03-26:
can you check the size of the firmware vs whatever is available on the mcu... unfortunately I do not have access to wroom_2

alinelena, 2024-03-26:
add the miso pin even if not used...

alinelena, 2024-03-26:
ESP8266 I am sure will not have enough memory do you have by any chance an esp32 or something else to try?

deltamelter, 2024-03-26:
> ESP8266 I am sure will not have enough memory do you have by any chance an esp32 or something else to try?

only some esp32-cam modules with sdcard and camera, will need to work out which pins are free.  
They use a lot of power and run hot even when idle but does seem to run without the EPD connected.

I'm waiting on some esp32 versions of these battery boards, the idea is to use them as deep-sleeping devices that only update the panel once or twice a day.

alinelena, 2024-03-26:
it is just for a test, so shall not matter. 
personally I am using this https://www.aliexpress.com/item/1005004960064227.html but I am not using batteries... https://gitlab.com/m-lego/sensor rev1 working on rev2

deltamelter, 2024-03-26:
Not many free pins on the esp32-cam and with the camera "flash" led tied to gpio04, the room lit up every time the panel was update, but it updated without problems.    I only had to change 1 pin (DC->gpio00 to gpio12)
I think the esp-wroom-02 must be conflicting pin usage, I will try to find an alternative while I wait for ESP32 boards.
The GPIO16 is soldered/connected to GND for deep sleep to be able to wake.  I wonder if this might cause some problems? 

btw, is there already something to work with `model: 2.13in3c`?

alinelena, 2024-03-27:
> btw, is there already something to work with `model: 2.13in3c`?

yes, unfortunately there is mess in the sense someone opened another similar feature request #2191 and info is scattered now... 

now I will try to summarise, for the 3 sizes we act has and colour i have.:
- [x] bw 2.13 works no issues with esphome [1]

```yaml
   221 display:                                                                                                                            
   222   - platform: waveshare_epaper                                                                                                      
...
   225     model: "2.13inv3" 
```

- [x] bw 2.9 works with minimal changes, see a draft in here, please note this needs some work to abstract it properly to be merged. https://github.com/esphome/esphome/pull/6217 see [2]

- [x] bw 1.54 works no issues with esphome [3]
```yaml
 238 display:                                                                                                                            
   239   - platform: waveshare_epaper  
...                                                                                                    
   241     model: "1.54inv2"  
```

- [ ] bw+red 2.9 works with  https://github.com/esphome/esphome/pull/6226

- [ ] bw+red 2.13 reported to work but I did not check... a quick look at 2.9 pr tells me one needs only to change the height from 296 to 250, the rest of the code shall be identical. in this way you will avoid the issue reported in here https://github.com/esphome/feature-requests/issues/2191#issuecomment-2009352588 

[1][image]

[2] [image]

[3] [image]

deltamelter, 2024-03-27:
Is it specific to this panel that it pushes an update as soon as the display initialises?  Always starts up before the other sensors have been set leaving `nan` and "placeholder" values, then after first subsequent update, the panel shows actual data.  I don't want to waste this cycle.  How can the initial update be triggered only when data is present?

alinelena, 2024-03-27:
> Is it specific to this panel that it pushes an update as soon as the display initialises? Always starts up before the other sensors have been set leaving `nan` and "placeholder" values, then after [...]

depends. always you can use some interval, i sugges to join esphome discord and ask for help in there about this.

mstsclev, 2025-01-14:
work in Waveshare 4,2 rev v2.2 if set 2.90in3c and github://pr#6226. but not full screen. How to solve this problem? Is it possible to make support for large screens too?
[image]