esphome/feature-requests #2102: Olimex ESP32-POE(-ISO) Ethernet & Wifi at the same time

sipodd, 2023-02-05:
**Describe the problem you have/What new integration you would like**

Possibility to enable Ethernet, Wifi and Bluetooth, on this particular Boards at the same time.

**Please describe your use case for this integration and alternatives you've tried:**

While the Hardware seems to be capable, it is not possible to enable Ethernet and Wifi on Olimex ESP32-POE at the same time.

**Additional context**

I would like to use Olimex ESP32-POE Boards as Wifi AP while the devices are connected and powered via POE-Ethernet. (so: using Ethernet as an uplink)

nagyrobi, 2023-02-05:
This is not specific to Olimex ESP32-POE. ESPHome doesn't support parallel usage of Ethernet and Wifi, regardless of the board:
https://esphome.io/components/ethernet.html

sipodd, 2023-02-05:
> This is not specific to Olimex ESP32-POE. ESPHome doesn't support parallel usage of Ethernet and Wifi, regardless of the board: https://esphome.io/components/ethernet.html

I know, thats the reason for this FR...i'd like to have it supported.

JeroenVanOort, 2023-02-13:
Using an ESP32 as an ethernet backed wifi access point would require more than just being able to use the two components simultaneously, as traffic would need to be routed between the two interfaces. Although I couldn't help you with the specifics because I simply don't know, I believe it would take quite some work to add this to esphome.

Out of curiosity, I looked for ESP32 based AP's outside of esphome and while there are some projects, I don't have the impression that your use case is a common one. From this I take that people have found other solutions to be better suited. Unless you want a hobby project to spend many hours on, I would advise you to buy a (second hand) AP and use a [PoE splitter](https://letmegooglethat.com/?q=poe+splitter+gigabit+12v). From the 12V jack, you can (using some DIY) feed both the AP and the ESP32. Most AP's also serve as an ethernet switch, so the ESP could take it's connectivity from there. This is what I use for PoE powering some (OpenWrt based) AP's that don't support PoE directly.

mdrobniu, 2023-07-12:
Is there a way just only to enable wifi association. i.e. I'd like to open my garadge door once esp home completes a 4-way handshake with my carplay.

kpfleming, 2023-07-29:
I too would like to use both Ethernet and WiFi simultaneously, but not as an AP or bridge... just regular 'client' connectivity on both interfaces.

ssieb, 2023-09-29:
> +1 Bluetooth and Ethernet at the same time needed

@Grandma-Betty that's already possible.  This is about wifi.

nagyrobi, 2023-09-29:
Works for me with the same board... what Bluetooth devices are you trying to use?

nagyrobi, 2023-09-29:
> Are you using the same code for ethernet?

Yes, and use a static IP address with `manual_ip` and `use_address` for higher stability.

ssieb, 2023-09-29:
If you use `manual_ip`, you don't need `use_address`.

kpfleming, 2023-09-29:
I suspect it does not need to be 'synchronized to an NTP server', it just needs to have the wall-clock time set in some fashion. Using the `homeassistant` time platform should be sufficient as well.

Gunni, 2026-09-14:
Why not openthread too? Or a generic networking route table?