esphome/feature-requests #2105: Add support for Mitsubishi Heavy Duty Industries ZS-W

diogocardoso28, 2023-02-05:
**Describe the problem you have/What new integration you would like**

Add support for Mitsubishi Heavy Duty Industries ZS-W on remote climate.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to control my ac from home assistant. I have tried heatpump_IR but it doesnt support my unit. 

**Additional context**

I have a remote data dump from heatpump_IR:
################# Start

Number of symbols: 155
Symbols:
Hh01001010011101011100001101011000101001111101111100100000010111111010000000111111110000001011011101001000111111110000000011111111000000001111111000000001
Bytes:
00: 0100|1010 | 52 | 01010010
01: 0111|0101 | AE | 10101110
02: 1100|0011 | C3 | 11000011
03: 0101|1000 | 1A | 00011010
04: 1010|0111 | E5 | 11100101
05: 1101|1111 | FB | 11111011
06: 0010|0000 | 04 | 00000100
07: 0101|1111 | FA | 11111010
08: 1010|0000 | 05 | 00000101
09: 0011|1111 | FC | 11111100
10: 1100|0000 | 03 | 00000011
11: 1011|0111 | ED | 11101101
12: 0100|1000 | 12 | 00010010
13: 1111|1111 | FF | 11111111
14: 0000|0000 | 00 | 00000000
15: 1111|1111 | FF | 11111111
16: 0000|0000 | 00 | 00000000
17: 1111|1110 | 7F | 01111111
18: 0000|0001 | 80 | 10000000
52,AE,C3,1A,E5,FB,04,FA,05,FC,03,ED,12,FF,00,FF,00,7F,80
Timings (in us):
PAUSE SPACE: 0
HEADER MARK: 3328
HEADER SPACE: 1424
BIT MARK: 512
ZERO SPACE: 239
ONE SPACE: 1036
Decoding known protocols...
Looks like a Mitsubishi Heavy ZM-S protocol (Model SRKxxZM-S Remote Control RLA502A700B) or
looks like a Mitsubishi Heavy ZS-S protocol (Model SRKxxZS-S Remote Control RLA502A700L)
POWER OFF
MODE HEAT
Temperature: 22
FAN 3
Vertical air direction: SWING
Horizontal air direction: SWING
3D Auto: ON
Night setback: OFF
Silent mode: OFF
Clean or Alergen: OFF
################# End

RoboMagus, 2023-02-07:
Have you looked at [this repo](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome)?

I've had hit and miss experiences with IR devices in the past, and having ESPHome directly communicate with the AC unit both greatly improves reliability and provides device state feedback when interacting with the device outside of HomeAssistant / ESPHome.

It does require you to open up the AC unit though!

BrianLanghoor, 2024-05-18:
I have the same request. Got a SCM60ZS-W unit with a RLA502A704A remote controller. Have no idea how diogo got to ready the bytes. So I'm hoping they are the same as mine.